cocotb/cocotb #1499: Test names are incorrect in summary

ktbarrett, 2020-03-15:
From https://github.com/cocotb/cocotb/pull/1380#issuecomment-599155133. Introduced in #1262. It's due to all the unnecessary monkeying the decorators and regression manager do to compute a name. The PR changed how the module name is passed to `_store_test_result`. It uses `__module__`, but on `RunningTest`s that value is `cocotb.decorator` instead of the module name of the underlying function.

ktbarrett, 2020-03-15:
`RunningTest` mostly exists for use by the regression manager. It really doesn't need all the extra attributes on it, they already exist on the `test` object. The regression manager should be changed to just work off `test` objects instead of stuffing all the info it needs from that object into yet another object. `RunningTest` and `RunningCoroutine` should just go away entirely. I think this will require a change to the scheduler as well as the regression manager due to how `handle_result` works. The relationship between the scheduler and regression manager are just very very irregular...

eric-wieser, 2020-03-15:
To be clear, you're advocating that `RunningTest` and `RunningCoroutine` should both be replaced with their `RunningTask` baseclass, which doesn't have any of the weird attributes.

eric-wieser, 2020-03-15:
> The relationship between the scheduler and regression manager are just very very irregular...

What I'd eventually like to happen is for the regression manager to just be a single coroutine run by the scheduler, with the weirdness wrapped into a few new scheduler methods or context managers

ktbarrett, 2020-03-15:
@eric-wieser Yes, deprecated `RunningTest` and `RunningCoroutine` and change the decorators to create `RunningTasks` in their place. The regression manager could simply `fork` a test "watcher", which in turn `fork`s the test coroutine and records the outcome of the test once it finishes. I don't think we need any additional scheduler features to support that.

eric-wieser, 2020-03-15:
We need a feature to collect and kill all coroutines forked by the test before starting the next one, which is currently embedded in the scheduler.

ktbarrett, 2020-03-15:
I think we should destroy the scheduler between tests and have each trigger/coroutine clean itself up. But don't we already have `finish_scheduler` for that purpose?

PR #1515 (2020-03-20): Fix test name in regression summary
Fix for regression introduced by 1abdd1d12f049415743f6fcc6b617228d2f85bcb which caused the incorrect test name to be displayed.

Fixes #1499. The refactor discussed will occur in a separate PR.
commit 815b0cb9 (2020-03-20): Fix test names in regression summary (#1515)

Fix for a regression introduced by 1abdd1d which caused an incorrect test name containing `cocotb.decorator` to be displayed.

Fixes #1499.
closed 2020-03-20 by eric-wieser