cocotb/cocotb #1500: SINGLETON_HANDLES define

ktbarrett, 2020-03-15:
What is it? What is it trying to solve? Does it work? It either needs documentation, or a PR to remove it and associated code (`CLEAR_STORE`).

garmin-mjames, 2020-03-16:
It stores a map of `path name` -> `GpiObjHdl*` so that subsequent accesses to an object don't have to do the full conversion from native (VPI/VHPI/FLI) to GPI objects each time.
I've never tested performance with it on/off.

ktbarrett, 2020-03-16:
@garmin-mjames Enforcing single handles to objects actually seems actively bad. VHDL has the concept of aliases. Aliases likely would resolve to the same object and have different paths requiring different handles, right?

garmin-mjames, 2020-03-16:
I think it would depend on the alias, as you can alias a sub-range of a vector for instance. The alias or parts of the alias would reference the same simulator objects behind the scenes. As far as I understand it, it doesn't necessarily mean the object handles from `VHPI` will be the same.

The context of the commit that added this (7ede134a3bd750f8d7857f9ff1773a88867e9db9) is iteration over a Python handle #17. I'm not sure why it is/was necessary since `vpi_scan`/`vhpi_scan` should give you objects without duplication.

ktbarrett, 2020-03-24:
I think I might see what this was for. VPI object lifetimes *should* be tied to Python handles, but they are not. It is possible when referencing the same object more than once that new VPI objects keep getting allocated and never freed. This acts as a cache to prevent that.

eric-wieser, 2020-03-24:
I've some local changes which might be relevent to that end, but they're gated on #1427 being merged.

garmin-mjames, 2020-05-28:
VHPI and VPI iterators make use of the "caching" functionality when creating pseudo-regions. The pseudo-region object re-uses the handle of the parent region object that contains the generator statement.

https://github.com/cocotb/cocotb/blob/924f35a3b7d39543118b7bfaed77dd4808e6612b/cocotb/share/lib/vhpi/VhpiCbHdl.cpp#L1152-L1165

https://github.com/cocotb/cocotb/blob/924f35a3b7d39543118b7bfaed77dd4808e6612b/cocotb/share/lib/vpi/VpiCbHdl.cpp#L854-L867

ktbarrett, 2020-07-14:
The `SINGLETON_HANDLES` cache's cleanup behavior causes use-after-free. See https://github.com/cocotb/cocotb/pull/1782#issuecomment-627739972 for details.

ktbarrett, 2020-07-14:
`SINGLETON_HANDLE` will be very difficult to replace. It is used to ensure the lifetime of all handles is the entire program, and that one cannot obtain multiple copies of the same handle to prevent memory leaks due to the unimplemented resource collection (`gpi_free_handle`).

The alternative is have handles allocated only while in use and deallocate them after use. Because there can be multiple users of a handle at any time, handles will have to be ref-counted. To ensure that we always obtain the same subhandle from a handle so ref-counting works, we must adopt a tree-like structure. Subhandles (children) will link to handles (parents) to keep parents alive as long as they have children alive. Drivers and Loads form cross-branch references, but this isn't a big deal, as the data structure cannot form a cycle due to how HDLs work (we could still catch pathological cases). Linking handle lifetime in C++ to handle lifetime in Python will need research (I know pybind11 handles this, that might be a starting place for inspiration).

As you gathered, this is a rewrite of most of the GPI. Something similar to this, a tree structure of handles, was considered in #826 but for other reasons.

ktbarrett, 2020-11-17:
The `SINGLETON_HANDLES` cache and the object lifetime system should be documented in the internal documentation.

closed 2026-01-12 by ktbarrett