cocotb/cocotb #1503: Add deprecation warning to `BinaryValue.value`

ktbarrett, 2020-03-17:
The doctsring mentions this property is deprecated. Perhaps we should try to enforce that. A `DeprecationWarning` should be issued, and all usages of `BinaryValue.value` should be wiped from the tests and replaced with `BinaryValue.integer` or `BinaryValue.signed_integer`, or whatever is appropriate.

eric-wieser, 2020-03-17:
It's a bit weird how the signed and unsigned integer setters are equivalent.

Given the known issues with `BinaryValue` today, I'm inclined to not add any deprecation warnings until we have a goal design in sight - we don't want to deprecate `.value` only to add it back again.

Thanks for filing this separately though. Can we find the commit that decided to deprecate these?

ktbarrett, 2020-03-17:
Lol, it's been deprecated since the beginning of time. 7380e94202e33f3ee7a3369da0096b13560fce1d. I'm with you that I don't really care much about deprecating `.value` now if we are going to replace or refactor `BinaryValue` soon.

eric-wieser, 2020-03-19:
Let's leave this open, but not do anything till we have a clearer picture.

ktbarrett, 2023-12-13:
Will be removed by #3244.

closed 2023-12-13 by ktbarrett