cocotb/cocotb #1504: Refactor of handle objects

ktbarrett, 2020-03-17:
Handles (in handle.py) support dot syntax for traversal of the HDL object hierarchy, but they also contain a number of attributes for using the value. These namespaces are known to collide and there currently is no mechanism to work around this issue. And to some degree it is unavoidable, Python objects have a number of builtin attributes (usually starting with 2 underscores).

To solve this I recommend we split the traversal and analysis of HDL objects into two. Object traversal objects will only be able to be used to traverse the hierarchy. We can then obtain a `HDLValue` objects which will contain all the other expected and useful behavior. One would obtain this value with an attribute that will not collide with valid identifiers in Verilog, VHDL or Python: `use` is a keyword in both languages and not a keyword in Python.

```python
type(dut.input)  # -> child of SimHandleBase
int(dut.input)  # -> Error
type(dut.input.use)  # -> HDLValue
int(dut.input.use)  # -> 0
```

Alternatively, since we should already mention Python builtin attributes names (`__{name}__`) will not work we could put such a function in a similar name (e.g. `__value__`), and provide a free function for access.

```python
type(HDLValue(dut.input))  # -> HDLValue
int(HDLValue(dut.input))  # -> 0
```

eric-wieser, 2020-03-17:
Can you give some concrete examples of name conflicts, and how each example would be spelt under each of your proposed new spellings?

eric-wieser, 2020-03-17:
(in particular, is it only `.value` that is problematic?)

ktbarrett, 2020-03-18:
I have run into `.value` before, it's the biggest offender. In that case the solution was to rename the object :/ This may not be easy depending upon the user. Formal verification processes are extremely bureaucratic, turn around time on a rename might be over a week. But *every* attribute on those classes is a valid Verilog identifier (VHDL identifiers can't start with `_`) and has the potential to collide.

* `setimmediatevalue`, `drivers`, `loads`, and `value`: valid Verilog and VHDL identifiers
* `_thing`: while they are "private", they are still valid Verilog identitifers
* `__operator__`: still valid Verilog identifiers, but unfortunately unavoidable

I just made this issue to highlight an implementation flaw that might one day be worked and poll for potential solution.

ktbarrett, 2020-03-18:
Another potential solution is to make indexes always do hierarchical lookup, but allow dot syntax to remain. Indexes will *always* be HDL traversal only and there can be no collisions, but dot syntax may collide. If you ever run into a name collision, just switch syntaxes, or just always use indexing.

```python
type(dut["value"])  # -> SimHandleBase
type(dut.value)  # -> property
```

eric-wieser, 2020-03-18:
> But every attribute on those classes is a valid Verilog identifier

This is only actually an issue for heirarchical objects though, right? Is it possible for an object to both have both heirarchical descedents, and also have meaningful `drivers`, `loads`, `values` etc?

To be clear, I'm not disagreeing with you - I just want to end up with a precise description of the problem here.

eric-wieser, 2020-03-18:
> __operator__: still valid Verilog identifiers, but unfortunately unavoidable

Not true, because python only cares about `type(dut).__operator__`, we're free to replace `dut.__operator__` via `dut.__getattribute__`.

closed 2024-12-22 by ktbarrett