cocotb/cocotb #1505: Provide Verilog Sources in a .vf file

Alex-Mann, 2020-03-18:
Hi there,

I am wondering if it is possible to provide a `design.vf` file which is generated and contains all the paths to all the design files needed for compilation. For example, this would appear in the makefile in the `axi_lite_slave` example as:
```
// design.vf
axi_defines.v
axi_lite_demo.v
axi_lite_slave.v
tb_axi_lite_salve.v
```

And then in the makefile, you would include such a file like so:
```
// hdl/Makefile
VERILOG_SOURCES = $(TOPDIR)/hdl/design.vf
```

I have tried this but it does not compile properly as it throws a syntax error.

eric-wieser, 2020-03-18:
Are you looking for the following?
```makefile
VERILOG_SOURCES = $(shell cat $(TOPDIR)/hdl/design.vf)
```

Alex-Mann, 2020-03-18:
Yeah so my Makefile fu isn't the greatest... I've tried that though and this is the error I get when I run `make`
```
make[1]: Entering directory `/cocotb/tests'
mkdir -p sim_build
make[1]: *** No rule to make target `axi_lite_slave.v', needed by `sim_build/sim.vvp'.  Stop.
make[1]: Leaving directory `/cocotb/tests'
make: *** [sim] Error 2
```

eric-wieser, 2020-03-18:
Yeah, you probably need to prepend the folder name. My makefile fu is also bad. Since this is at this point a make question ("How can I turn this file into this variable value") and not a cocotb one, I'd recommend asking on stack overflow.

When you find a solution, it might be a nice thing to include as an example somewhere.

closed 2020-11-24 by github-actions[bot]