cocotb/cocotb #1506: Illegal option -o pipefail

silvan-hdl, 2020-03-18:
Hi

I try to bring Cocotb to work with Modelsim from Intel.
Modelsim works fine. When running the Makefile, it breaks with the following message:

make[2]: Verzeichnis „/home/silvan/Documents/Arbeit/HDL/ct_modelsim/env/lib/python3.7/site-packages/cocotb/share/lib/fli“ wird verlassen
set -o pipefail; cd sim_build && LD_LIBRARY_PATH=/home/silvan/Documents/Arbeit/HDL/ct_modelsim/cocotb/examples/dff/tests/build/libs/x86_64::/usr/lib:/usr/lib:/usr/lib:/usr/lib MODULE=dff_cocotb TESTCASE= TOPLEVEL="work.dff" COCOTB_SIM=1 \
GPI_EXTRA= TOPLEVEL_LANG=verilog PYTHONPATH=/home/silvan/Documents/Arbeit/HDL/ct_modelsim/cocotb/examples/dff/tests/build/libs/x86_64:/home/silvan/Documents/Arbeit/HDL/ct_modelsim/cocotb/examples/dff/tests:/home/silvan/Documents/Arbeit/HDL/ct_modelsim/env/lib/python3.7/site-packages: \

/opt/intelFPGA_pro/19.4/modelsim_ase/bin/vsim -c -64  -do runsim.do 2>&1 | tee sim.log
/bin/sh: 1: set: Illegal option -o pipefail

Running echo $SHELL returns /bin/bash
The first line in vsim is #!/bin/sh

System: Linux Ubuntu 19.10 x64

Any ideas?
Thank you for your help

imphil, 2020-03-18:
This sounds like a dash issue. Can you try adding `SHELL=bash` to `Makefile.questa`? Can you also check what shell `/bin/sh` is linked to?

cmarqu, 2020-03-19:
Related: https://github.com/cocotb/cocotb/pull/897#discussion_r279330205

silvan-hdl, 2020-03-19:
Thank you imphil
Thank you cmarqu

adding SHELL=bash solved the problem.

cmarqu, 2020-03-29:
Let's leave this open until we have a pull request that sets ``SHELL`` for all simulators.
Unfortunately, doing this in ``Makefile.sim`` breaks things for me in a weird way... I'm still investigating.

cmarqu, 2020-03-29:
> Unfortunately, doing this in `Makefile.sim` breaks things for me in a weird way... I'm still investigating.

This turned out to be a bug in the makefiles for the Cadence simulators.

PR #1544 (2020-03-29): Set SHELL=bash for the pipefail option used in many simulator Makefiles
Use bash for the ``pipefail`` option used in many simulator Makefiles.
Move setting ``pipefail`` before defining the envvars to make for Cadence simulators.

Closes #1506.
closed 2020-03-29 by cmarqu