cocotb/cocotb #1509: The performance between python and SV

zlzg12345, 2020-03-19:
Hi,  
We use SV for our projects now, but we are interested in cocotb becuase python has large library and support running without  compile.  
We write some examples to check the performance beteen cocotb and pure sv and find it takes much more time (about 5 times,  we expected it just a littler slower),. We are not sure if we make any mistake when we use cocotb. 

Is there anything to speed up (or slow)  the simulation when use cocotb ? 
Did you analyze the performance between cocotb and pure SV enviroment ? if you did, could you let me know the results ?

Thanks
Le

eric-wieser, 2020-03-19:
Are you using `yield` or `await`?

zlzg12345, 2020-03-19:
using yield.

imphil, 2020-03-19:
It's really hard to give specific advice here without knowing more about your setup and your code. Can you post somewhere (a) what simulator you're using (including the version and operating system), (b) what code you're running, and (c) what method for profiling you're using.

eric-wieser, 2020-03-19:
You'll get a small performance boost by switching to `async` / `await`, especially if your test-cases call lots of nested coroutines. If you perform lots of value assignments, there's a patch in the works that will greatly speed up that too.

zlzg12345, 2020-03-20:
> It's really hard to give specific advice here without knowing more about your setup and your code. Can you post somewhere (a) what simulator you're using (including the version and operating system), (b) what code you're running, and (c) what method for profiling you're using.

Hi,
Thanks for the reply.  We are using : 
a). vcsmx/2017.12_sp1  & Red Hat Enterprise Linux Server release 6.10.
b). The simple driver (I2S,  one audio interface, very simple and common) we wrote base on your example, it is used to replace the existed pure SV i2s driver in our legacy testbench (It still include many other drivers in SV).
c), We run the same testcase (only the drivers are different, one is python, the other one is SV), then 
 check how long it takes.

zlzg12345, 2020-03-20:
> You'll get a small performance boost by switching to `async` / `await`, especially if your test-cases call lots of nested coroutines. If you perform lots of value assignments, there's a patch in the works that will greatly speed up that too.

Thanks for the reply. I will try "async and await",   and  where can I find the patch ?

themperek, 2020-03-20:
Async support: https://cocotb.readthedocs.io/en/latest/coroutines.html#async-generators

Depending on how you start simulator the difference may come from optimization settings (`acc`). I do not know how this works in VCS.

eric-wieser, 2020-03-20:
The patch I'm referring to is gh-1286, which is waiting on gh-1502, which is waiting on gh-1507, which is nearly ready.

cmarqu, 2020-03-20:
> Depending on how you start simulator the difference may come from optimization settings (`acc`). I do not know how this works in VCS.

Good point, the two simulations may use different settings there.
VCS also has some kind of "access learning mode" where it records which hierarchical elements it needs access to, and in a second simulation it can read that information and optimize out unneeded accesses to elements.

cmarqu, 2020-05-01:
See also #1676.

closed 2020-11-24 by github-actions[bot]