cocotb/cocotb #1512: Improve error reporting for `yield` inside async function

ktbarrett, 2020-03-19:
Those that are in the process of converting `cocotb.coroutine` functions using `yield` to async functions may run into problems where they miss a `yield`, which turns their coroutine into an async generator. If an ill-formed coroutine makes it to the scheduler it will result in a less than helpful error message. Instead we should check for async generators where we expect coroutines and throw a `TypeError` a more helpful message, much like we do with old style coroutines missing `yield` statements.

eric-wieser, 2020-03-19:
Good idea

eric-wieser, 2020-03-19:
I assume the current error message is from these lines:

https://github.com/cocotb/cocotb/blob/c4a573defa921e5a16e63f585481ad3790bcf5d7/cocotb/decorators.py#L90-L98

Is all you're suggesting a more precise error if `inspect.isasyncgen` is true?

ktbarrett, 2020-03-19:
@eric-wieser I think that will be the only place that needs to change, `fork`s and `yield`s must go through that, when using `await` you have different semantics. But I will test.

PR #1516 (2020-03-20): Better error messages when accidentally using yield in coroutines
Fixes #1512. There are a few points where we expect coroutines, so each place got a warning.
commit 022880ee (2020-03-20): Improve error message when accidentally using async generators. (#1516)

Fixes #1512.

For users converting `cocotb.coroutine` functions using `yield` to async functions using `await`, this provide a better error message when they forget a `yield` and accidentally turn their coroutine into an async generator.
closed 2020-03-20 by eric-wieser