cocotb/cocotb #1527: Makefile.ghdl: some improvements

csantosb, 2020-03-22:
Hi,

  Running here latest cocotb from git under ArchLinux, doing a vhdl simulation with help of latest ghdl-llvm. 

  As you may see
 
[image]

running `make` twice, without any source code modification, does `ghdl -i` followed by `ghdl -m`, which is unnecessary, especially when handling hundreds of vhdl source files.

I have tried to imporve things by tunning `Makefile.ghdl`, result is [here](https://github.com/csantosb/cocotb/commit/a8ce22c05a4ceec34f392967ab02f9a148820d93). Now, calling `make` twice again gives

[image]

Only `ghdl -r` is run now, as one would expect.

Realize that, when a single vhdl file is modified, the former behaviour holds, which is still incorrect: only the concerned source file should be analyzed upon calling `ghdl -m`, no need to include it back again. Also, I had to use a new variable `GHDL_STD`, taking `08`, `93c` or similar.

eric-wieser, 2020-03-22:
Sounds like the issue is that `analyze` is a phony target, so make is not able to work out when it needs to be rerun.

csantosb, 2020-03-22:
Just checked. Commenting out the PHONY line makes no difference.

eric-wieser, 2020-03-22:
The problem isn't that we've labelled it `.PHONY`, it's that it _is_ phony because it doesn't actually create the target that it says it does.

How did you deduce the name of the `.cf` file that is created? Can we specify it explicitly?

csantosb, 2020-03-22:
As for the `.cf`, [here](https://ghdl.readthedocs.io/en/latest/using/ImplementationOfVHDL.html#library-database) you have the details.

Currently, `--work` stands for `$(RTL_LIBRARY)`. Concerning the `GROUP` (vhdl standard), there is not yet an equivalent variable. Notice how [workdir](https://ghdl.readthedocs.io/en/latest/using/InvokingGHDL.html#cmdoption-ghdl-workdir) equals `$(SIM_BUILD)` too.

tgingold, 2020-04-14:
I fear that the change can only deal with the work library.
Maybe instead of -i/-m, the -c/-e options should be tried.
Or do a touch after elaboration that only depends on the VHDL sources.

csantosb, 2020-04-14:
As for right now, below is the last up to date version of `Makefile.ghdl` which I'm using to deal with this issue.

Far from perfect, but otherwise I have to re-elaborate the whole thing upon every modification to python testbench. You are right, it only deals with work lib, and also includes non-cocotb variables.

I'll try to explore the alternative you're proposing.

```
.PHONY: analyse

$(SIM_BUILD)/$(RTL_LIBRARY)-obj$(GHDL_STD).cf: $(VHDL_SOURCES)
	mkdir -p $(SIM_BUILD)
	$(foreach SOURCES_VAR, $(filter VHDL_SOURCES_%, $(.VARIABLES)), \
	$(CMD) -i $(GHDL_ARGS) $(COMPILE_ARGS) --work=$(SOURCES_VAR:VHDL_SOURCES_%=%) --workdir=$(SIM_BUILD) $($(SOURCES_VAR)) && ) \
	$(CMD) -i $(GHDL_ARGS) $(COMPILE_ARGS) --work=$(RTL_LIBRARY) --workdir=$(SIM_BUILD) $(VHDL_SOURCES)

$(SIM_BUILD)/$(TOPLEVEL): $(SIM_BUILD)/$(RTL_LIBRARY)-obj$(GHDL_STD).cf $(CUSTOM_COMPILE_DEPS)
	$(CMD) -m -o $(SIM_BUILD)/$(TOPLEVEL) $(GHDL_ARGS) $(COMPILE_ARGS) --work=$(RTL_LIBRARY) --workdir=$(SIM_BUILD) $(TOPLEVEL)

# Compilation phase
analyse: $(SIM_BUILD)/$(RTL_LIBRARY)-obj$(GHDL_STD).cf

$(COCOTB_RESULTS_FILE): $(SIM_BUILD)/$(TOPLEVEL) $(CUSTOM_SIM_DEPS)
	cd $(SIM_BUILD) && \
	rm -f $(COCOTB_RESULTS_FILE) && \
	PYTHONPATH=$(PWD):$(PYTHONPATH) MODULE=$(MODULE) \
	TESTCASE=$(TESTCASE) TOPLEVEL=$(TOPLEVEL) TOPLEVEL_LANG=$(TOPLEVEL_LANG) COCOTB_SIM=1 \
	$(CMD) -r $(GHDL_ARGS) $(TOPLEVEL) --vpi=$(LIB_DIR)/libcocotbvpi_ghdl.$(LIB_EXT) $(SIM_ARGS) $(PLUSARGS)

    $(call check_for_results_file)

clean::
	-@rm -rf $(SIM_BUILD)
```

csantosb, 2020-04-26:
I gave a permanent home to my custom Makefile.ghdl file.

    https://gitlab.com/ip-vhdl/eda-common/-/blob/master/cocotb/Makefile.ghdl

Hope it helps.

cmarqu, 2020-04-26:
I guess we all feel a bit reluctant with this PR because our CI for ghdl is not good - we allow it to fail and still let Travis report an overall PASS currently.
Since ghdl is at 0.37 since a few months, it's worth taking another look at its CI.

csantosb, 2020-04-27:
Well, I just intend to recall that when one modifies the test bench only (python code), the make command re-elaborates the whole again and again, which is weird, specially for large projects. For sure, as it is, my custom makefile is not good, specially because I need to update it continuously. It just fits me.