cocotb/cocotb #1533: Splitting up test_cocotb

ktbarrett, 2020-03-23:
The test_cocotb module in `/tests/test_cases/test_cocotb/` is currently an assortment of tests to test various features, but it's been kinda of a dumping ground. We also have test projects for testing bugs that are usually tied to issue numbers, and projects for testing various features (like test_iteration_vhdl). So it seems like test_cocotb exists to test the core functionality of cocotb, mainly the scheduler and adjacent functionality.

That being said, I think we should probably do a cleanup pass on the issue test projects and test_cocotb to align them according to the above. But we should also split test_cocotb up into several modules according to category, because it's getting to be a rather large collection of tests.

garmin-mjames, 2020-03-23:
Are project maintainers opposed to using `pytest` or `unittest` to test some of the `Python`-only code?
For instance I have some local `BinaryValue` unit tests I wrote with `pytest` (`BinaryValue` is so broken...)

eric-wieser, 2020-03-23:
I would be very in favor of adding some `pytest` tests to CI for the python-only code.

ktbarrett, 2020-04-05:
So far I am thinking splitting the tests along these lines:
* deprecation checking (makes a good spot to know all the code we have deprecated)
* async generators (don't work in 3.5 so they have to be split off)
* generator-based coroutines
* async coroutines
  * tests for valid coroutine checking
  * tests for coroutine behavior (propagating exceptions, killing, etc)
  * tests for concurrency
* Python triggers
* GPI triggers

ktbarrett, 2020-04-10:
Closed in #1594.

closed 2020-04-10 by ktbarrett