cocotb/cocotb #1539: Quickstart fails on Centos 7.7

gchqdev1729, 2020-03-26:
Following the Quckstart instructions in README.md failed for me. I'm using CentOS 7.7 with Python 3.6.8. I think swig needs to be added to the list of prerequisites for Linux. It also doesn't seem to work with swig 2.0.10 - maybe this is expected?

I would supply a patch to the docs, but I'm uncertain exactly what the dependency should be so that it works on all systems. I only have my CentOS box and I'm new to swig.

Without swig installed the tests fail with:

```
     0.00ns CRITICAL cocotb.regression                         regression.py:153  in initialise                      Failed to import module test_endian_swapper_hal: No module named 'hal'
```

It took me longer than it should have to notice that further up it said that it could not find swig:

```
swig -python -outcurrentdir ../hal/endian_swapper_hal.h
make[2]: swig: Command not found
make[2]: *** [endian_swapper_hal_wrap.c] Error 127
make[2]: Leaving directory `/home/user/git/cocotb/examples/endian_swapper/cosim'
make[1]: [hal] Error 2 (ignored)
```

With swig 2.0.10 installed it does better, but still fails:

```
30755.03ns ERROR    cocotb.regression                         regression.py:396  in _score_test                     Test Failed: initial_hal_test (result was SystemError)
                                                                                                                     StopIteration
                                                                                                                     
                                                                                                                     The above exception was the direct cause of the following exception:
                                                                                                                     
                                                                                                                     SystemError: <built-in function delete_endian_swapper_state_t> returned a result with an error set
```

With package swig3 (3.0.12) installed all is well.

themperek, 2020-03-26:
Can you try some other example like adder.
 This particular example is not so easy quick/actually. 

@eric-wieser Maybe we should enable hal example only in travis/ci ?

gchqdev1729, 2020-03-26:
Adder works fine right out of the box.

eric-wieser, 2020-03-26:
@gchqdev1729, does this patch make it fail loudly?
```
--- C:\Users\wiese\Repos\cocotb\examples\endian_swapper\tests\Makefile	Wed Mar 11 21:57:02 2020
+++ C:\Users\wiese\Repos\cocotb\examples\endian_swapper\tests\Makefile	Thu Mar 26 17:46:33 2020
@@ -69,7 +69,7 @@
 ifeq ($(OS),Linux)
 .PHONY: hal
 hal:
-	-cd ../cosim && make
+	cd ../cosim && make
 
 clean::
 	-cd ../cosim && make clean
```

gchqdev1729, 2020-03-27:
Yes, it does. Without swig installed at all:

```
(venv) [user@centos tests]$ make
make results.xml
make[1]: Entering directory `/home/user/git/cocotb/examples/endian_swapper/tests'
mkdir -p sim_build
cd ../cosim && make
make[2]: Entering directory `/home/user/git/cocotb/examples/endian_swapper/cosim'
gcc  -fPIC -pthread -fno-strict-aliasing -O2 -pipe -Wall -Wp,-D_FORTIFY_SOURCE=2 \
-fexceptions --param=ssp-buffer-size=4 -D_GNU_SOURCE \
-fwrapv -DNDEBUG -I/usr/include/python3.6m -c io.c -o io_module.o
gcc -pthread -shared  io_module.o -L/usr/lib64  -o io_module.so
swig -python -outcurrentdir ../hal/endian_swapper_hal.h
make[2]: swig: Command not found
make[2]: *** [endian_swapper_hal_wrap.c] Error 127
make[2]: Leaving directory `/home/user/git/cocotb/examples/endian_swapper/cosim'
make[1]: *** [hal] Error 2
make[1]: Leaving directory `/home/user/git/cocotb/examples/endian_swapper/tests'
make: *** [sim] Error 2
```

Much clearer. I would definitely have spotted that right away. :)

swig 2 vs swig 3 remains a problem, of course, but perhaps I would have known that if I knew more about swig?

themperek, 2020-03-27:
@eric-wieser Should we maybe move to something like pybind11 and setuptools? I can put this on my list.

eric-wieser, 2020-03-27:
Let's just fix the makefile for now. It's only an example, it doesn't really matter how it builds its extension module.

Protoplasmoloid, 2020-03-28:
Thanks Eric for putting me here.  
If I run this using WSL+Ubuntu both examples work OK.  
If I run them using VBox+Ubuntu (Both 18.04LTS, and with swig3.0.12-1) I get for endian-swapper:
     0.00ns CRITICAL cocotb.regression                         regression.py:153  in initialise                      Failed to import module test_endian_swapper_hal: No module named 'hal'
And for adder:
test -f results.xml
/home/git/cocotb/cocotb/share/makefiles/simulators/Makefile.ghdl:68: recipe for target 'results.xml' failed
make[1]: *** [results.xml] Error 1
make[1]: Leaving directory '/home/git/cocotb/examples/adder/tests'
/home/git/cocotb/cocotb/share/makefiles/Makefile.sim:150: recipe for target 'sim' failed
make: *** [sim] Error 2
That latter did run the simulation however, so I think the issue here is something else.
Hope you can help, and I can be of assistance :-)

eric-wieser, 2020-03-28:
@Protoplasmoloid, would you mind submitting a PR with the suggested one-letter makefile change in the linked issue? I'm on mobile so can't fix it right now.

Protoplasmoloid, 2020-03-28:
@eric-wieser Sorry, just a noob here wanting to learn VHDL.  PR=pull request? I get that, but no idea about your request. Sorry again!

eric-wieser, 2020-03-29:
@Protoplasmoloid: can you try again with the latest master?

cmarqu, 2020-03-29:
There actually was something reported/fixed for Questa in #897 that looks related.
@gchqdev1729, @Protoplasmoloid - are you using Questa?

Protoplasmoloid, 2020-03-29:
I am not using Questa, just GHDL. Did pull from Master as you asked...
Now for Ububtu18.04+WSL I get for endian-swapper and adder:

$ make SIM=ghdl TOPLEVEL_LANG=vhdl
make: cocotb-config: Command not found
Makefile:66: /Makefile.inc: No such file or directory
make: cocotb-config: Command not found
Makefile:67: /Makefile.sim: No such file or directory
make: *** No rule to make target '/Makefile.sim'.  Stop.

The above did work OK before, although I want to run in a VM.

and for Ububtu18.04 in VBox for endian-swapper I still get:

     0.00ns CRITICAL cocotb.regression                         regression.py:153  in initialise                      Failed to import module test_endian_swapper_hal: No module named 'hal'
 
and still for Adder, it passes tests, but at the end:

# check that the file was actually created, since we can't set an exit code from cocotb
test -f results.xml
/home/git/cocotb/cocotb/share/makefiles/simulators/Makefile.ghdl:68: recipe for target 'results.xml' failed
make[1]: *** [results.xml] Error 1
make[1]: Leaving directory '/home/git/cocotb/examples/adder/tests'
/home/git/cocotb/cocotb/share/makefiles/Makefile.sim:153: recipe for target 'sim' failed
make: *** [sim] Error 2

Thanks.

cmarqu, 2020-03-29:
I have no idea about the WSL setup (I suggest opening a new issue), but for your other setup:

It seems like the Makefile for GHDL doesn't respect ``CUSTOM_COMPILE_DEPS``.
Untested: you could change line 60 in ``cocotb/share/makefiles/simulators/Makefile.ghdl`` so that it looks as follows:
```
analyse: $(CUSTOM_COMPILE_DEPS) $(VHDL_SOURCES) $(SIM_BUILD)
```

Currently, it is just
https://github.com/cocotb/cocotb/blob/9c3773580ab52c6e72a88d718b3742c99b11530e/cocotb/share/makefiles/simulators/Makefile.ghdl#L60

Protoplasmoloid, 2020-03-30:
@cmarqu .  Thank you, updated Makefile, now get:
     0.00ns CRITICAL cocotb.regression                         regression.py:153  in initialise                      Failed to import module test_endian_swapper_hal: /home/git/cocotb/examples/endian_swapper/tests/../cosim/io_module.so: undefined symbol: PyInt_AsLong

I am using python3.8-dev.

eric-wieser, 2020-03-30:
That looks like you built `io_module.so` with python 2, then tried to import it with python 3.8. Perhaps `make clean` and rebuilding will fix that.

Protoplasmoloid, 2020-03-30:
You are quite right, I didn't think to do make clean in the endian-swapper, works fine now. Thank you.

eric-wieser, 2020-03-30:
@themperek, swapping the build of `io_module.so` to work through setuptools would give a unique name containing the python version number, which would avoid that problem. Not high priority though.

gchqdev1729, 2020-03-31:
> There actually was something reported/fixed for Questa in #897 that looks related.
> @gchqdev1729, @Protoplasmoloid - are you using Questa?

Sorry for the silence... I mainly use Modelsim, but in this case I'm just on open source tools.

cmarqu, 2020-04-04:
> You are quite right, I didn't think to do make clean in the endian-swapper, works fine now.

@Protoplasmoloid Did it also fix your WSL+Ubuntu problem?

Protoplasmoloid, 2020-04-05:
@cmarqu For some reason the Ubuntu+WSL always did work (and initially with Python2, before I did a pull), just the vanilla Ubuntu in VBox didn't, both 18.04LTS. Oh, and both Python3. Thank you.

eric-wieser, 2020-04-09:
I'm going to close this, I think we fixed all the issues that we diagnosed. If you still have problems on master, please open a new issue!

Thanks for working through this with us, @gchqdev1729 and @Protoplasmoloid.

PR #1543 (2020-03-29): Do not ignore fail in hal target
This is to "fail early" and prevents a more obscure fail later in the run.
Fixes #1539.
commit f5dbd831 (2020-03-29): Do not ignore fail in hal target (#1543)

This is to "fail early" and prevents a more obscure fail later in the run.
Fixes #1539.
closed 2020-04-09 by eric-wieser