cocotb/cocotb #1547: Consider using a stricter flake8 configuration

eric-wieser, 2020-03-30:
Opening this to track a todo in the source

https://github.com/cocotb/cocotb/blob/244ae8f3c903e7f90953d1deae83687e1d622af0/setup.cfg#L10-L34

This list of ignores was populated based on what was already violated, not based on any strong opinions about what was right.

ktbarrett, 2020-03-30:
Should this also include cleaning up the lines where `noqa` is used to suppress those warnings? For example, it's used to prevent complaining when aligning code, which is mostly aesthetic.

eric-wieser, 2020-03-30:
`noqa` is usually a deliberate choice, much of the contents of the flake8 configuration is accidental - so I don't think this issue needs to cover `noqa` cleanup, nor do I think it's a problem if the resolution is to remove some ignores at the expense of (very few) `# noqa`s.

cmarqu, 2020-05-02:
To make this process easier, the following line will help (note that the ignore list doesn't contain E302, E303, and E305): 
``autopep8 -i -r --max-line-length 200 --ignore E123,E126,E127,E128,E131,E202,E203,E221,E225,E226,E228,E231,E241,E402,E501,E741,F403,F405,F841,W504 <dirs_to_run_on>``

NOTE: do **not** run autopep8 with the full list enabled, as that would destroy some manual alignment etc. we want to keep.

eric-wieser, 2020-05-03:
@cmarqu: sounds like you've already made progress on this, want to submit a PR with what you've done so far?

cmarqu, 2020-05-03:
Will do soon.

cmarqu, 2020-05-11:
The remaining ignores seem to be intentional, so we can close this (and #6) after all PRs are merged.

andruvictor, 2020-05-21:
try this http://joker123gaming.co/

eric-wieser, 2020-05-21:
I think we might want to fix the `F841`

cmarqu, 2020-07-10:
> I think we might want to fix the `F841`

Many of those are in tests and examples where it's mostly that return values get a nice name for didactic purposes.
I'll use flake8's ``per-file-ignores`` for everything in these directories and fix the ones in core code.

ktbarrett, 2025-02-04:
After introducing ruff, the majority of these have been removed except those pylint warnings / style issues about too many returns, too many branches, and other subjective issues like that.

closed 2025-02-04 by ktbarrett