cocotb/cocotb #1548: Some escaped identifiers containing '.' are not accessible from cocotb

thasti, 2020-03-30:
Verilog's escaped identifiers can start with a backslash and then contain any printable ASCII character, while ending with a space. Such identifiers are often found in netlists, for example after unrolling a generate loop.

In the following simple example:
```
module test(
);

  wire \with0.special_name = 0;
  wire \with1.special_name = 0;
  wire \with2.special_name = 0;

endmodule
```
Only the last of these three definitons is accessible.

Even after doing `dut._discover_all()`, `dut._sub_handles` contains only one element

```
{'special_name ': ModifiableObject(test.\with2.special_name )}
```

All three handles are shadowed onto 'special_name' it appears.

eric-wieser, 2020-04-08:
Do any log messages appear when starting the simulation?

thasti, 2020-04-08:
No log messages appear (verified against current master). I did some superficial digging and found this: https://github.com/cocotb/cocotb/blob/master/cocotb/handle.py#L215 as a potential origin of the problem.

eric-wieser, 2020-04-08:
Yeah, I just found that too. Made a draft patch, but don't have the time to put together a test case.

PR #1610 (2025-01-31): Fix handling of escaped identifiers containing dots
Needs tests.

Fixes #1548, probably.
closed 2025-01-31 by ktbarrett