cocotb/cocotb #1550: Normalize and improve type checking and support where coroutines are expected

ktbarrett, 2020-03-30:
There are many places that currently expect coroutines:

- `First`, `Combine`, `with_timeout`
- `yield` statements in old-style coroutines
- `fork`

All these places should be able to accept async def `coroutine`s, or `RunningTask`s, and much like `fork` do type checking. `First`, `Combine` and `with_timeout` don't currently accept `coroutine`s, and everything in the list has slightly different type checking. This can all be done in a central place by using a function that does the type checking and turns coroutine-like types into `RunningTask`s.

Additionally, but slightly differently, `RunningTask`'s constructor expects `coroutine` or generators and should trap common errors as well.

I think this is as simple as moving all the types checking to `RunningTask` and implementing a function that tries to convert the value if it is not already a `RunningTask`.

eric-wieser, 2020-05-04:
Learnings from #1752 are that we need a special error message for when `inspect.isasyncgenfunction(coro)`, because users are likely to forget to convert a yield to an await

ktbarrett, 2020-05-04:
My intention is to move all type checking into `cocotb.decorators.coroutine` for coroutine functions and `cocotb.decorators.RunningTask` for coroutine objects. I will also add in #1593 to the constructors. Then I will introduce.

```python
def make_coroutine(thing) -> cocotb.decorators.coroutine:
    if isinstance(thing, cocotb.decorators.coroutine):
        return thing
    else:
        return cocotb.decorators.coroutine(thing)

def make_task(thing) -> cocotb.decorators.RunningTask:
    if isinstance(thing, cocotb.decorators.RunningTask):
        return thing
    else:
        return cocotb.decorators.RunningTask(thing)

def iscoroutinefunction(thing) -> bool:
    return not throws(make_coroutine, thing)

def iscoroutine(thing) -> bool:
    return not throws(make_task, thing)

def isyeildable(thing) -> bool:  # name?
    """ Schedulable thing that can go in a yield/await statement in cocotb """
    return iscoroutine(thing) or isinstance(thing, (cocotb.triggers.Trigger, cocotb.triggers.Waitable))
```

These can be used all over: `scheduler.add`'s checks can be replaced with `make_task`, an `iscoroutine` check can be done in #1752. There are a lot of places in the codebase that do very similar type checking but do it to varying degrees of effectiveness. We shoudl locate and replace them with these functions.

eric-wieser, 2020-05-04:
I don't think we ever need the result of `make_coroutine` though, do we?

ktbarrett, 2020-05-04:
Probably not, I just included it for sake of completion.

closed 2024-12-22 by ktbarrett