cocotb/cocotb #1552: Do not append to make variables that we expect users to set from the command line

cmarqu, 2020-03-31:
When we tell users to use ``SIM_ARGS`` or ``EXTRA_ARGS`` etc. to add their options and expect that this works for them on the command line like make ``SIM_ARGS=-whatever``, we cannot append to these variables in our ``Makefile.<simulator>``, because only the variable values given on the command line would be set (https://stackoverflow.com/a/32696474).

This will be relevant for the documentation part of #1495.

Affected: 
* ``EXTRA_ARGS``  in ``Makefile.xcelium``/``Makefile.ius``, ``Makefile.vcs``
* ``COMPILE_ARGS`` in ``Makefile.verilator``
* probably more

eric-wieser, 2020-04-02:
Isn't the solution here to teach users to use `EXTRA_ARGS=... make` instead of `make EXTRA_ARGS=...`?

cmarqu, 2020-04-02:
Indeed it is (see also https://github.com/cocotb/cocotb/pull/1553#issuecomment-608074148 for reference).

I would have like to get around this because setting the environment variable is a shell-dependent, and thus a bit more complicated to document (e.g. fish and csh users need to use ``env EXTRA_ARGS=...``).

PR #1590 (2020-05-11): Mention proper way of setting make vars
Document that users should use the syntax ``EXTRA_ARGS=... make`` instead of ``make  EXTRA_ARGS=...`` because some Makefiles append to these variables internally, and the second syntax would break them.

Closes #1552
commit cacc21f8 (2020-05-11): Mention proper way of setting make vars

Document that users should use the syntax
EXTRA_ARGS=... make
instead of
make EXTRA_ARGS=...
because some Makefiles append to these variables internally,
and the second syntax would break them.

Closes #1552
closed 2020-05-11 by cmarqu