cocotb/cocotb #1566: Remove COCOTB_SIM environmental variable

themperek, 2020-04-04:
Since https://github.com/cocotb/cocotb/pull/1425 it should not be needed

ktbarrett, 2020-04-06:
I am fine with removing this as an environment variable, however I'd like some way for a user to know if they operating in a valid cocotb context or not. If `simulator` can be loaded unconditionally, perhaps we can put a function there to disambiguate (`running` property)? Or maybe the function that I will soon add that gets the product and version strings could return "nosim" if no simulator is running?

The reason I ask is because I am trying to write a compatibility layer for concurrency primitives so we can run our models for cocotb tests in python-only contexts using asyncio. I need to know what context the library is running in to know whether to source cocotb or asyncio implementations.

eric-wieser, 2020-04-22:
As an interim solution, could we remove this today, and change `_initialize_testbench` to do the following?
```
IN_SIMULATOR = false

def _initialize_testbench(...):
    nonlocal IN_SIMULATOR
    IN_SIMULATOR = true
```

themperek, 2020-04-22:
I am for any solution.

Maybe add `get_sim_name()` to siumlator module? Retun `None` if none?

themperek, 2020-04-22:
So if one does:

``python3
SIM_NAME = None

def _initialize_testbench(...):
..
```

Should be able to check for `cocotb.SIM_NAME` if I am not mistaken?

eric-wieser, 2020-04-22:
My suggestion above was an attempt to continue avoiding importing the simulator module, without the need for an env var.

We can't solve that problem from the simulator until we solve #1568 (or re-introduced `SPHINX_BUILD` as a variable)

themperek, 2020-04-22:
@eric-wieser We cannot import `cocotb.simulator` now? I just cannot call some functions but able to import.

eric-wieser, 2020-04-22:
That only works on systems with `--enabled-shared`. Ideally we'd lift that restriction by reversing the dependency between `libgpi` and `libcocotb`.

ktbarrett, 2020-04-22:
@eric-wieser Funnily enough, I have that change hanging out as well. I was hoping to take a swing at refactoring the scheduler before that, but I can move up those changes first so they can be done over the next few weeks.

themperek, 2020-04-22:
@eric-wieser 
>That only works on systems with --enabled-shared. Ideally we'd lift that restriction by reversing the >dependency between libgpi and libcocotb.

Any case without `--enabled-shared` setup will not work and something will fail since we need to link with python?

eric-wieser, 2020-04-22:
@themperek: Not if users just want to inspect / mock / call the simulator API. The methods could all `raise RuntimeError("No GPI backend")`.

themperek, 2020-04-22:
@eric-wieser So you see a way to remove the dependency on linking pythonXY.so during setup for all libraries?

eric-wieser, 2020-04-22:
No, my proposal would be to not build `libcocotb` under the circumstances where there is no python.so, but arrange the dependencies so that everything else can be built.

themperek, 2020-04-22:
>No, my proposal would be to not build libcocotb under the circumstances where there is no python.so

Why one would do this? Would this not add even more confusion later on? Not better do #1618 ?

ktbarrett, 2020-04-22:
@themperek there is a possibility that someone want to wrap the GPI in a language other than Python. While cocotb is currently the way we distribute the GPI, it would be nice if we didn't shackle it to Python. I'm not speaking for Eric, but that would be my reasoning.

themperek, 2020-04-26:
Actually any kind of solution like proposed in  https://github.com/cocotb/cocotb/issues/1566#issuecomment-617644287 have looks not working straightforward. One needs to know if is in the simulation during cocotb module loading (before we can set anything) due to :
https://github.com/cocotb/cocotb/blob/a3fb4b559da411ecc863e26a84c2948bc71de67b/cocotb/__init__.py#L58
and than:
https://github.com/cocotb/cocotb/blob/a3fb4b559da411ecc863e26a84c2948bc71de67b/cocotb/log.py#L56
...
https://github.com/cocotb/cocotb/blob/a3fb4b559da411ecc863e26a84c2948bc71de67b/cocotb/utils.py#L54

One would need to protect this. Maybe try..except?

ktbarrett, 2020-04-26:
@themperek All of that code should be moved into `_initialize_testbench`. See #1332 for example of what that might look like. No code should be run during package loading IMO.

themperek, 2020-04-26:
So the some idea, for the momemt. Add this:
```c
_putenv_s("COCOTB_SIM", "1");
```
someware here:
https://github.com/cocotb/cocotb/blob/a3fb4b559da411ecc863e26a84c2948bc71de67b/cocotb/share/lib/gpi/GpiCommon.cpp#L208-L210

Can you remove from makefiles and do something better letter?

ktbarrett, 2020-04-26:
@themperek I'd prefer to do it the right way the first time. I can write up a PR right now for just that, though it will probably conflict with #1647.

ktbarrett, 2020-04-26:
My change works... once the simulator module is built. However, because the change makes the cocotb runtime import the simulator module unconditionally, and because the build script is located *in* the cocotb package we end up trying to import the cart before the horse. Perhaps the version module as well.

eric-wieser, 2020-04-27:
As of #1713, this variable is no longer set in makefiles, and is instead set here:

https://github.com/cocotb/cocotb/blob/ad977c1dd26569a9360fc478567ec0595bd8f0a0/cocotb/share/lib/gpi/GpiCommon.cpp#L211-L214

We should aim to replace these lines with something less hacky - let's leave this issue open for now.

PR #1717 (2020-05-14): Unconditional import of cocotb and removal of COCOTB_SIM
Closes #1566. This removes `COCOTB_SIM` from the cocotb Python package. It was typically used to condition import of the simulator module. There was a number of accesses on the simulator module that caused segfaults if it wasn't conditioned. Now,

- the simulator module is imported unconditionally (usage still causes a segfault). Supplants #1568.
- all code that accessed the simulator module that ran at cocotb module's load has been moved so it runs later, in code paths only used for actual simulations
- because of that unconditional imp [... cut]
closed 2020-05-14 by imphil