cocotb/cocotb #1567: Verilator can export .vcd file but the documentation doesn't mention that

minecraft2048, 2020-04-05:
In documentation at https://cocotb.readthedocs.io/en/latest/building.html#var-WAVES it says that the `WAVES` make variable can be enabled to enable vcd export for  Aldec Riviera-PRO and Mentor Graphics Questa simulators, but it doesn't mention Verilator and it doesn't work for iverilog. But when I googled cocotb verilator I found this: https://antmicro.com/blog/2019/06/verilog-with-cocotb-and-verilator/ then it says that we can set `VERILATOR_TRACE` make variable to make Verilator export .vcd file, then when I try it, it works.

I think there are 2 ways to fix the documentation:

1. Say in the documentation that `VERILATOR_TRACE` can be used to enable wave dump for Verilator, or
2. Make so that `WAVES` variable work for Verilator

Martoni, 2020-04-08:
we can use [EXTRA_ARGS with --trace option.](https://github.com/cocotb/cocotb/pull/1607)
But If WAVES be supported it be best yes :)

cmarqu, 2020-04-08:
Actually, we plan to deprecate such simulator-specific variables and replace them with documentation on how to get the same effect.
Regarding ``WAVES``, I'm still undecided whether it should get the same treatment. Personally, I find at least the name misleading, and it's not clear what to do if not all signals should be dumped, or how to specify at what hierarchy level(s), or at what simulation times etc.

bluecmd, 2020-09-08:
This seems to be fixed these days: https://docs.cocotb.org/en/stable/simulator_support.html#sim-verilator-waveforms

cmarqu, 2020-09-08:
@bluecmd Thanks for the heads up.

closed 2020-09-08 by cmarqu