cocotb/cocotb #1569: Installation does not work on readthedocs

eric-wieser, 2020-04-05:
If I check this checkbox:

[image]

Then the build fails with:
```
building 'cocotb/libs/libgpilog' extension
creating build/temp.linux-x86_64-3.7/cocotb/share/lib/gpi_log
gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -Icocotb/share/include -I/home/docs/.pyenv/versions/3.7.3/include/python3.7m -c cocotb/share/lib/gpi_log/gpi_logging.cpp -o build/temp.linux-x86_64-3.7/cocotb/share/lib/gpi_log/gpi_logging.o -std=c++11 -Wall -Wextra -Wcast-qual -Wwrite-strings -Wconversion -Wnon-virtual-dtor -Woverloaded-virtual -D__STDC_FORMAT_MACROS
g++ -pthread -shared -L/home/docs/.pyenv/versions/3.7.3/lib -L/home/docs/.pyenv/versions/3.7.3/lib build/temp.linux-x86_64-3.7/cocotb/share/lib/gpi_log/gpi_logging.o -Lbuild/lib.linux-x86_64-3.7/cocotb/libs -lpython3.7m -lcocotbutils -o build/lib.linux-x86_64-3.7/cocotb/libs/libgpilog.so
/usr/bin/ld: /home/docs/.pyenv/versions/3.7.3/lib/libpython3.7m.a(ceval.o): relocation R_X86_64_PC32 against symbol `_PyRuntime' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: final link failed: Bad value
collect2: error: ld returned 1 exit status
error: command 'g++' failed with exit status 1
```

I'm a little worried this represents a regression WRT #1183. @tymonx, would you be able to test that master still works for your use-case?

themperek, 2020-04-05:
I think we are following #1183. Is this python compiled with "--enable-shared"?

eric-wieser, 2020-04-05:
I think you're right that it isn't compiled with `--enable-shared`. Wasn't the purpose of #1183 to make it work in that case though?

themperek, 2020-04-05:
I am not sure but as I understand it was to fix if you have both.

eric-wieser, 2020-04-05:
If you have both `--enable-shared` and what?

themperek, 2020-04-05:
Maybe check what `python -c "import sysconfig;print(sysconfig.get_config_vars())"` gives?

themperek, 2020-04-05:
>If you have both

Static and dynamic?

eric-wieser, 2020-04-05:
> import sysconfig;print(sysconfig.get_config_vars())

Good thinking, done here: https://readthedocs.org/projects/cocotb/builds/10770164/

eric-wieser, 2020-04-05:
Likely relevant: https://stackoverflow.com/questions/46469587/readthedocs-sphinx-build-failing-but-working-locally

eric-wieser, 2020-04-05:
Looking more closely, it seems the `libpython.so` file is simply absent from readthedocs.

closed 2020-04-29 by eric-wieser