cocotb/cocotb #1580: "Running tests with cocotb ... from Unknown"

cmarqu, 2020-04-05:
In e.g. https://asciinema.org/a/316949, we can see the line
```
0.00ns INFO  cocotb  __init__.py:144  in _initialise_testbench  Running tests with cocotb v1.4.0.dev0 from Unknown
```

@themperek suggested using ``__file__`` in https://github.com/cocotb/cocotb/blob/09effc69f914820d4aa50217b9fad6ba29c4248f/cocotb/__init__.py#L139-L144 instead of ``export``ing ``COCOTB_PY_DIR`` in https://github.com/cocotb/cocotb/blob/67bf79b997b630416b1f98742fbeab09d9e90fef/cocotb/share/makefiles/Makefile.inc#L33

eric-wieser, 2020-04-05:
How did it end up as unknown in the first place? Is that variable just not used any more?

themperek, 2020-04-05:
@eric-wieser https://github.com/cocotb/cocotb/pull/1445

Should add `export` or use`__file__` ?

eric-wieser, 2020-04-05:
Using `__file__` sounds like a good idea to me. Hadn't noticed the lack of `export`.

PR #1581 (2020-04-05): Fix "from Unknown" in the log
Resolves #1580
closed 2020-04-05 by eric-wieser