cocotb/cocotb #1582: Add type annotations for user-facing code

eric-wieser, 2020-04-06:
Now that Python 3.5 is our minimum supported version, we should consider adding type annotations to public methods and functions.

Split from #1289 since this is more of an enhancement than a cleanup.

ktbarrett, 2020-04-28:
What is considered an appropriate amount of annotation? Are we annotating public functions for the sake of documentation only? Or are we adding annotations so that we may lint the codebase at some point? The latter will require more extensive annotations. Once we get to a level of correctness we could add a mpypy step to the linting stage of the CI.

eric-wieser, 2020-04-28:
My vote would be err towards documentation first, and we can add the linting if we care later.