cocotb/cocotb #1585: Remove support for cross-compilation in the endian example

eric-wieser, 2020-04-06:
We have some weird code here that seemingly tries to support building 32-bit extension modules with 64-bit python:

https://github.com/cocotb/cocotb/blob/688d391376eda47446d8f949d9e62f8ca93d51b0/cocotb/share/makefiles/Makefile.pylib#L34-L35

https://github.com/cocotb/cocotb/blob/688d391376eda47446d8f949d9e62f8ca93d51b0/cocotb/share/makefiles/Makefile.pylib.Linux#L34-L38

https://github.com/cocotb/cocotb/blob/688d391376eda47446d8f949d9e62f8ca93d51b0/cocotb/share/makefiles/Makefile.pylib.Darwin#L34-L38

I don't know why we'd ever want to support this. If the user needs a 32-bit extension module, then they should be using `cocotb-config` installed inside a 32-bit python.

If we removed it, we could merge all the `Makefile.pylib` files into a single file

themperek, 2020-04-06:
I am for it. In any case, it may not be so easy and not sure that those lines will help (rather more confusion). 

It can be moved to the endian example worst case.

eric-wieser, 2020-04-06:
> It can be moved to the endian example worst case.

Or perhaps also in the best case. After removing these workaround, there are only two interesting lines left in the makefile:

https://github.com/cocotb/cocotb/blob/688d391376eda47446d8f949d9e62f8ca93d51b0/cocotb/share/makefiles/Makefile.pylib.Linux#L32

https://github.com/cocotb/cocotb/blob/688d391376eda47446d8f949d9e62f8ca93d51b0/cocotb/share/makefiles/Makefile.pylib.Linux#L40

These could easily be moved to within the endian example makefile

eric-wieser, 2020-04-10:
After gh-1619, this is already not supported - we may as well delete the makefiles now.

PR #1632 (2020-04-15): Remove unused Makefile.pylib files.
After #1306 and #1619 those files are not used anymore. 

Resolves #1585
closed 2020-04-15 by eric-wieser