cocotb/cocotb #1591: Problems with MODULE

ktbarrett, 2020-04-07:
If the `MODULE` env variable is not set cocotb fails hard (raises an exception in the startup routine). It seems to me it should just run and exit with no failure.

`MODULE` might also accidentally contain a trailing or leading separators, this isn't handled, and tests show it fails in an ugly way. `HOOKS`, whose usefulness is debatable, suffers the same problem.

ktbarrett, 2020-04-07:
We should also have a tests that ensures things still work when we have `MODULE`, but don't have any tests.

ktbarrett, 2020-04-07:
Also, if the elements are not stripped it could fail. It would be easy enough to add.

eric-wieser, 2020-04-07:
> If the MODULE env variable is not set cocotb fails hard by raising (an ImportError of all things...) in the startup routine. It seems to me it should just run and exit with no failure because there are no tests to run.

I don't agree, if it's not set I then I think we should fail. The behavior you describe seems appropriate if it's deliberately set to an empty string though.

eric-wieser, 2020-04-07:
> `MODULE` might also accidentally contain trailing or leading separators, this isn't handled, and tests show it fails in an ugly way.

This sounds like a good idea to me

imphil, 2020-04-07:
@ktbarrett can you outline your use case for "MODULE" being emtpy, and the reason why this should not be an error? From a usability perspective I think an error message about a missing/empty MODULE variable with a hint why cocotb doesn't run anything?

ktbarrett, 2020-04-07:
I think most people consider the current behavior as odd. Imagine if pytest ran no tests because there were none specified or found and it failed hard?

Specifying test modules according to conditions is not an unreasonable use case. And you may end up with a set of conditions that lead to not tests being run.

ktbarrett, 2020-04-07:
Whoops, using a phone sucks.

We can meet in the middle and turn that error into a warning level log.

ktbarrett, 2020-04-07:
I guess I missed Eric's concern. What exists now does not support an empty, but existant, `MODULE`. That may also be a good middle ground. 

So error if not set, warn if empty, but pass?

eric-wieser, 2020-04-07:
> So error if not set. Warn if empty, but pass?

That sounds sensible to me.

eric-wieser, 2020-04-07:
> Imagine if pytest ran no tests because there were none specified or found and it failed hard?

Pytest _does_ do this!

ktbarrett, 2020-04-07:
@eric-wieser It warns, not errors. But the error code is non-0 so you could consider it a failure.

eric-wieser, 2020-04-07:
> But the error code is non-0 so you could consider it a failure.

Indeed, that was the point I was making.

imphil, 2020-04-07:
Following pytest is generally a good thing, so let's do that. Warn + non-zero, but distinguishable exit code (5 in pytest's case). And please document that. 

For everyone else following along: that's what pytest shows:

```
$ pytest
==================================================================================== test session starts ====================================================================================
platform linux -- Python 3.8.2, pytest-5.4.1, py-1.8.1, pluggy-0.13.1
rootdir: /empty/root/dir
plugins: timeout-1.3.4, cov-2.8.1
collected 0 items                                                                                                                                                                           

=================================================================================== no tests ran in 0.35s ===================================================================================
$ echo $?
5
```

https://docs.pytest.org/en/latest/usage.html#possible-exit-codes

eric-wieser, 2020-04-07:
@imphil: cocotb doesn't have control of the exit code.

At any rate, passing `MODULE=""` should be treated the same as passing `MODULE="file_with_no_tests"`, which I think today is to pass.

ktbarrett, 2020-04-07:
@eric-wieser No, right now the behavior is fail hard and don't make a result.xml if fed the empty string. `''` in Python is `False`.

https://github.com/cocotb/cocotb/blob/afeb587cd239ad955860555a30f0f5add95f14bf/cocotb/__init__.py#L172-L174

imphil, 2020-04-07:
> @imphil: cocotb doesn't have control of the exit code.

Ah yes indeed.

So that leaves us with these options?

1. MODULE not set: ERROR (did we agree on that?)
2. MODULE set, but resolving (somehow) into a empty list of tests:
	a. WARN with non-zero exit code
    b. WARN with zero exit code
    c. ERROR

eric-wieser, 2020-04-07:
To be clear, my claim above is that today, `MODULE="file_with_no_tests"` passes with no warnings and emits an empty results.xml.

I think the thing to do is:

* Module not set: Error (like we do today, but with a clearer message)
* Tests resolve to an empty list: Emit `results.xml` as normal, print a warning.

ktbarrett, 2020-04-07:
Currently an unset `MODULE` is impossible in Icarus and probably most other simulators.

https://github.com/cocotb/cocotb/blob/afeb587cd239ad955860555a30f0f5add95f14bf/cocotb/share/makefiles/simulators/Makefile.icarus#L66-L68

PR #1592 (2020-04-08): Address issues with unexpected MODULE values
Fixes #1591. Also addresses issue with `HOOKS`.
closed 2020-04-08 by eric-wieser