cocotb/cocotb #1597: Support for regression-process listener

mballance, 2020-04-07:
I'm working on a randomization and functional coverage extension package, and I'd like to receive a callback when a regression ends so I can capture the regression status, simulation time, etc and write it to a coverage database.
In other unit-test frameworks (specifically googletest), I can add a listener to is notified when a test suite starts, when each test starts, when each test ends (and its status), and when the test suite/regression ends. I've looked and I don't see a way to do this with cocotb. Is there a way that I've overlooked?

Thanks in advance!

ktbarrett, 2020-04-21:
This will be easier to implement once we have a custom entry point in place.

mballance, 2020-04-21:
@ktbarrett, which issue/PR are you referring to with respect to a custom entry point? (just want to be sure I'm thinking the same direction you are).
Ideally I'd like to have this work a bit like a plug-in mechanism. Specifically, I'd like the act of Python loading certain "extension" packages to result in registering listeners with cocotb.

closed 2020-11-24 by github-actions[bot]