cocotb/cocotb #1601: Compatibility with Aldec Active-HDL on Windows category:windows category:simulators

pstrueb, 2020-04-07:
Dear cocotb community,

I'm new to cocotb and trying to get the endian_swapper exampl running.
My setup is: 
- Win10
- Aldec Active-HDL9.3 
- Miniconda with Python 3.7

What I did so far:
1. 1. Installed msys: conda install -c msys2 m2-base m2-make libpython m2w64-toolchain
2. 2. Install the development version of cocotb : 
3. - git clone https://github.com/cocotb/cocotb
4. - pip install -e ./cocotb
6. cd cocotb/examples/endian_swapper/tests
7. make SIM=aldec TOPLEVEL_LANG=vhdl

This leads to
```
runsim.tcl(5) error: parse error
@if [string length<--
/c/temp/cocotb/cocotb/share/makefiles/simulators/Makefile.aldec:144: recipe for target 'results.xml' failed
make[1]: *** [results.xml] Error 1
make[1]: Leaving directory '/c/temp/cocotb/examples/endian_swapper/tests'
c:\temp\cocotb\cocotb\share\makefiles/Makefile.sim:150: recipe for target 'sim' failed
make: *** [sim] Error 2
```
 So first of all it seems to that my vsimsa (build 9.3.2744.sp1.09.4995) cant handle the statements generated by Makefile.aldec
`@if [string length [array get env LICENSE_QUEUE]] {
 set LICENSE_QUEUE $::env(LICENSE_QUEUE)
}`
in runsim.tcl

So I removed in Makefile.aldec for testing:
        @echo "@if [string length [array get env LICENSE_QUEUE]] {" >> $@  
        @echo " set LICENSE_QUEUE $$::env(LICENSE_QUEUE)" >> $@  
        @echo "}" >> $@  

After running again with make Im getting:
`runsim.tcl(8) error: parse error
asim   +access +w -interceptcoutput -O2 -dbg -loadvhpi C:/temp/cocotb/cocotb/libs/libcocotbvhpi_aldec:vhpi_startup_routines_bootstrap<--
/c/temp/cocotb/cocotb/share/makefiles/simulators/Makefile.aldec:144: recipe for target 'results.xml' failed
make[2]: *** [results.xml] Error 1`

I could trace this problem back to vsimsa which seems to need "" around `-loadvhpi "C:/temp/cocotb/cocotb/libs/libcocotbvhpi_aldec:vhpi_startup_routines_bootstrap"`
After adding the quotation marks to Makefile.aldec:
```
else ifeq ($(TOPLEVEL_LANG),vhdl)  
    GPI_ARGS = -loadvhpi \"$(LIB_DIR)/libcocotbvhpi_aldec:vhpi_startup_routines_bootstrap\"
```
I again started with make. this now leads to execution of the generated script with vsimsa but Im getting `VHPI: Loading library 'C:/temp/cocotb/cocotb/libs/libcocotbvhpi_aldec.dll'
VHPI: Cannot load 'C:/temp/cocotb/cocotb/libs/libcocotbvhpi_aldec.dll' library. Library does not exist or is corrupted.`
The simulation seems to start running but never stops. 
This is how far I got with my Issue. I hope you can help me get cocotb running on my machine with this simulator and hopefully also improve the project also a bit.

eric-wieser, 2020-04-07:
Does `C:/temp/cocotb/cocotb/libs/libcocotbvhpi_aldec.dll` exist?

themperek, 2020-04-07:
I can only test Riviera so you have to help to figure this out. 
What version (32/64) you have of  Active-HDL9.3 and python?

themperek, 2020-04-08:
@pstrueb Do you have `mingw` directory in the installation?  Can you rename/remove it? Do you have other mingw/gcc/msys installed and in the `PATH`?

pstrueb, 2020-04-08:
> Does `C:/temp/cocotb/cocotb/libs/libcocotbvhpi_aldec.dll` exist?

I checked and yes, the .dll exists.

> I can only test Riviera so you have to help to figure this out.
> What version (32/64) you have of Active-HDL9.3 and python?

I have the 64 bit Version of python. As far as I see it, Active-HDL is a 32 bit application.

> @pstrueb Do you have `mingw` directory in the installation? Can you rename/remove it? Do you have other mingw/gcc/msys installed and in the `PATH`?

I checked my path and no other mingw/gcc/msys is there.  I also renamed the MINGW folder in Aldec\Active-HDL9.3. I had a MINGW folder directly located at C:\ which I also renamed but the outcome is still the same

themperek, 2020-04-08:
>I have the 64 bit Version of python. As far as I see it, Active-HDL is a 32 bit application. 

Do you use 32bit python/conda? example setup: https://github.com/cocotb/cocotb/wiki/Tier-2-Setup-Instructions#conda-based-installation-all-os

eric-wieser, 2020-04-08:
> runsim.tcl(5) error: parse error
> @if [string length<--

Is that `@` a typo? @themperek, does your version of aldec work without the `@`?

pstrueb, 2020-04-08:
I use 64 bit python/conda. Is it mandatory to use a 32 bit version ? I used 
`conda install -c msys2 m2-base m2-make libpython m2w64-toolchain`
for installing msys.

eric-wieser, 2020-04-08:
> Is it mandatory to use a 32 bit version

It's not possible to use 64-bit python from within a 32-bit simulator

pstrueb, 2020-04-08:
> > Is it mandatory to use a 32 bit version
> 
> It's not possible to use 64-bit python from within a 32-bit simulator

Okay I will try then again with a 32bit version of python.

themperek, 2020-04-08:
> Is that @ a typo in our makefile? @themperek, does your version of aldec work without the @?

It works in Riviera.

eric-wieser, 2020-04-08:
> > Is that @ a typo in our makefile? @themperek, does your version of aldec work without the @?
> 
> It works in Riviera.

Both with and without the `@`? If so, let's remove it - I don't see any evidence of it being legal TCL syntax.

themperek, 2020-04-08:
>Both with and without the @? If so, let's remove it - I don't see any evidence of it being legal TCL syntax.

Yes. Both. In case `@` seems to be just ignored.

I am not sure if not remove those `LICENSE_QUEUE` lines. This is user depend not sure if we should set anything here.

pstrueb, 2020-04-08:
I not set up a 32 bit python environment inside conda with 
```set CONDA_FORCE_32BIT=1    
conda create -n py37_32 python=3.7  
conda activate py3_32  
conda config --env --set subdir win-32  
conda install -c msys2 python=3 m2-base m2-make m2w64-toolchain libpython
```

After installing the development version again with pip, removing the `LICENSE_QUEUE` lines and adding the "", the vhpi libary loading error is gone.
Instead Im getting 

``` run -all  
     -.--ns INFO     cocotb.gpi                         ..mbed\gpi_embed.cpp:71   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     cocotb.gpi                         ..\gpi\GpiCommon.cpp:106  in gpi_print_registered_impl       VHPI registered
     0.00ns INFO     cocotb.gpi                                gpi_embed.cpp:332  in embed_sim_init                  Running on Aldec HDL Simulator version 9.03.2744
     0.00ns INFO     cocotb.gpi                                gpi_embed.cpp:333  in embed_sim_init                  Python interpreter initialized and cocotb loaded!
     0.00ns INFO     cocotb                                      __init__.py:145  in _initialise_testbench           RunTraceback (most recent call last): from c:\temp\cocotb32\cocotb

     0.00ns INFO     cocotb                                      __init__.py:162  in _initialise_testbench           Seeding Python random module with 1586339567
     0.00ns CRITICAL cocotb.regression                         regression.py:153  in initialise                      Failed to import module test_endian_swapper: No module named 'test_endian_swapper'    regression_manager.initialise()
     0.00ns INFO     cocotb.regression                         regression.py:154  in initialise                      MODULE variable was "test_endian_swapper"
  File "c:\temp\cocotb32\cocotb\regression.py", line 68, in _my_import
     0.00ns INFO     cocotb.regression                         regression.py:155  in initialise                      Traceback:
ModuleNotFoundError: No module named 'test_endian_swapper'
     0.00ns INFO     cocotb.regression                         regression.py:156  in initialise                      Traceback (most recent call last):
                                                                                                                       File "c:\temp\cocotb32\cocotb\regression.py", line 151, in initialisesys:1: ResourceWarning: unclosed file <_io.TextIOWra                                                                                                                         module = _my_import(module_name)
                                                                                                                       File "c:\temp\cocotb32\cocotb\regression.py", line 68, in _my_import
                                                                                                                         mod = __import__(name)
                                                                                                                     ModuleNotFoundError: No module named 'test_endian_swapper'

     0.00ns ERROR    cocotb.gpi                                gpi_embed.cpp:419  in embed_sim_init                  cocotb initialization failed - exiting
     0.00ns ERROR    cocotb.scheduler                            __init__.py:205  in _sim_event                      Failing test at simulator request before test run completion: Simulator shutdown prematurely```

eric-wieser, 2020-04-08:
Which commit of cocotb are you running? There was a very recent change to the aldec makefiles that corrected a regression with symptoms similar to that.

Can you paste the commit sha1 hash?

pstrueb, 2020-04-08:
Im on `commit 4c99ef423bf7b1fd815333e62c18043ebe311b8d`

eric-wieser, 2020-04-08:
Hmm, that's interesting. To help debug, could you edit `cocotb/__init__.py` in your installation, and make the following change:

```diff
--- C:\Users\wiese\Repos\cocotb\cocotb\__init__.py	Wed Apr  8 10:34:49 2020
+++ C:\Users\wiese\Repos\cocotb\cocotb\__init__.py	Wed Apr  8 11:25:59 2020
@@ -140,8 +140,8 @@
     if exec_path is None:
         exec_path = 'Unknown'
 
-    log.info("Running tests with cocotb v%s from %s" %
-             (__version__, exec_path))
+    log.info("Running tests with cocotb v%s from %s with cwd %s" %
+             (__version__, exec_path, os.getcwd()))
 
     # Create the base handle type
```
and show the result?

pstrueb, 2020-04-08:
Yes, I can try helping to debug.
After I made the changes suggested by you, the outcome is
```
run -all
     -.--ns INFO     cocotb.gpi                         ..mbed\gpi_embed.cpp:71   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     cocotb.gpi                         ..\gpi\GpiCommon.cpp:106  in gpi_print_registered_impl       VHPI registered
     0.00ns INFO     cocotb.gpi                                gpi_embed.cpp:332  in embed_sim_init                  Running on Aldec HDL Simulator version 9.03.2744Traceback (most recent call last):
     0.00ns INFO     cocotb.gpi                                gpi_embed.cpp:333  in embed_sim_init                  Python interpreter initialized and cocotb loaded!    if exec_path is None:
UnboundLocalError: local variable 'exec_path' referenced before assignment
     0.00ns ERROR    cocotb.gpi                                gpi_embed.cpp:419  in embed_sim_init                  cocotb initialization failed - exiting
     0.00ns ERROR    cocotb.scheduler                            __init__.py:204  in _sim_event                      Failing test at simulator request before test run completion: Simulator shutdown prematurely
sys:1: ResourceWarning: unclosed file <_io.TextIOWrapper name=1 mode='w' encoding='cp1252'> 
```

eric-wieser, 2020-04-08:
It sounds like you put my change in the wrong place, and added the context (gray) lines. You should have removed the red line, and added the green line.

pstrueb, 2020-04-08:
Yes you are right. The thing is that the line I should replace is 
```
 log.info("Running tests with cocotb v%s from %s" %
             (__version__, os.path.dirname(__file__)))
```

if i replace this with 
``` 
log.info("Running tests with cocotb v%s from %s with cwd %s" %
             (__version__, exec_path, os.getcwd()))
```
Im getting `UnboundLocalError: local variable 'exec_path' referenced before assignment` again which is understandable since exec_path is not declared.

eric-wieser, 2020-04-08:
Oh, looks like my version is out of date, apologies.

That last line should be `(__version__, os.path.dirname(__file__), os.getcwd()))`.

pstrueb, 2020-04-08:
Okay, that was some confusion but now its clear 👍 
The output now is 
```
run -all
     -.--ns INFO     cocotb.gpi                         ..mbed\gpi_embed.cpp:71   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     cocotb.gpi                         ..\gpi\GpiCommon.cpp:106  in gpi_print_registered_impl       VHPI registered
     0.00ns INFO     cocotb.gpi                                gpi_embed.cpp:332  in embed_sim_init                  Running on Aldec HDL Simulator version 9.03.2744
     0.00ns INFO     cocotb.gpi                                gpi_embed.cpp:333  in embed_sim_init                  Python interpreter initialized and cocotb loaded!
     0.00ns INFO     cocotb                                      __init__.py:145  in _initialise_testbench           Running tests with cocotb v1.4.0.dev0 from c:\temp\cocotb32\cocotb with cwd C:\temp\cocotb32\examples\endian_swapper\tests\sim_build
Traceback (most recent call last):
  File "c:\temp\cocotb32\cocotb\__init__.py", line 181, in _initialise_testbench
     0.00ns INFO     cocotb                                      __init__.py:161  in _initialise_testbench           Seeding Python random module with 1586345337
    regression_manager.initialise()
     0.00ns CRITICAL cocotb.regression                         regression.py:153  in initialise                      Failed to import module test_endian_swapper: No module named 'test_endian_swapper'  File "c:\temp\cocotb     0.00ns INFO     cocotb.regression                         regression.py:154  in initialise                      MODULE variable was "test_endian_swapper"
     0.00ns INFO     cocotb.regression                         regression.py:155  in initialise                      Traceback:
  File "c:\temp\cocotb32\cocotb\regression.py", line 68, in _my_import
     0.00ns INFO     cocotb.regression                         regression.py:156  in initialise                      Traceback (most recent call last):
ModuleNotFoundError: No module named 'test_endian_swapper'                                                             File "c:\temp\cocotb32\cocotb\regression.py", line 151, in initialise    mod = __import__(name)

                                                                                                                         module = _my_import(module_name)
                                                                                                                       File "c:\temp\cocotb32\cocotb\regression.py", line 68, in _my_importsys:1: ResourceWarning: unclosed file <_io.TextIOWrapper name=1 mode='w' encoding='cp1252'>

                                                                                                                         mod = __import__(name)
                                                                                                                     ModuleNotFoundError: No module named 'test_endian_swapper'

     0.00ns ERROR    cocotb.gpi                                gpi_embed.cpp:419  in embed_sim_init                  cocotb initialization failed - exiting
     0.00ns ERROR    cocotb.scheduler                            __init__.py:204  in _sim_event                      Failing test at simulator request before test run completion: Simulator shutdown prematurely
/c/temp/cocotb32/cocotb/share/makefiles/simulators/Makefile.aldec:144: recipe for target 'results.xml' failed
make[1]: *** [results.xml] Interrupt
c:\temp\cocotb32\cocotb\share\makefiles/Makefile.sim:150: recipe for target 'sim' failed
```

eric-wieser, 2020-04-08:
Thanks, that confirms my suspicion. Despite #1598, it seems that your cwd is `  C:\temp\cocotb32\examples\endian_swapper\tests\sim_build` instead of ` C:\temp\cocotb32\examples\endian_swapper\tests`.

eric-wieser, 2020-04-08:
Can you show the full output including the makefile commands?

pstrueb, 2020-04-08:
> Can you show the full output including the makefile commands?

Yes, sure: 
```
(py37_32) C:\temp\cocotb32\examples\endian_swapper\tests>make SIM=aldec TOPLEVEL_LANG=vhdl
make results.xml
make[1]: Entering directory '/c/temp/cocotb32/examples/endian_swapper/tests'
set -o pipefail; cd sim_build && GPI_EXTRA= TOPLEVEL_LANG=vhdl \
 MODULE=test_endian_swapper TESTCASE= TOPLEVEL=endian_swapper_vhdl COCOTB_SIM=1 /c/Aldec/Active-HDL9.3/BIN/vsimsa  -do runsim.tcl | tee sim.log

VHDL/Verilog/EDIF/SystemC Simulator build 9.3.2744.sp1.09.4995

(c) 1997-2014 Aldec, Inc. All rights reserved.

License Number 0

VSIMSA: Configuration files: `C:\Aldec\Active-HDL9.3\vlib\library.cfg', `C:\Aldec\Active-HDL9.3\BIN\vsimsa.cfg'
VSIMSA: `work' is now working library.
 Welcome to VSIMSA!
 This message was printed from `startup.do' macro file.

alib work
ALIB: Library `work' attached.
work = C:\temp\cocotb32\examples\endian_swapper\tests\sim_build\work\work.lib
set worklib work
VSIMSA: `work' is now working library.
acom  -dbg C:/temp/cocotb32/examples/endian_swapper/tests/../hdl/endian_swapper.vhdl
File: C:\temp\cocotb32\examples\endian_swapper\tests\..\hdl\endian_swapper.vhdl
Compile Entity "endian_swapper_vhdl"
Compile Architecture "impl" of Entity "endian_swapper_vhdl"
Compile success 0 Errors 0 Warnings  Analysis time :  0.2 [s]
asim   +access +w -interceptcoutput -O2 -dbg -loadvhpi "C:/temp/cocotb32/cocotb/libs/libcocotbvhpi_aldec:vhpi_startup_routines_bootstrap" endian_swapper_vhdl
VSIM: Selected architecture `impl' of entity `endian_swapper_vhdl' from library `work'.
ELBREAD: Elaboration process.
ELBREAD: Elaboration time 0.1 [s].
KERNEL: Main thread initiated.
KERNEL: Kernel process initialization phase.
KERNEL: Time resolution set to 1ps.
ELAB2: Elaboration final pass...
ELAB2: Create instances ...
KERNEL: PLI/VHPI kernel's engine initialization done.
PLI: Loading library 'C:\Aldec\Active-HDL9.3\bin\systf.dll'
PLI: Loading library 'C:\Aldec\Active-HDL9.3\bin\fsdbwriter.dll'
VHPI: Loading library 'C:/temp/cocotb32/cocotb/libs/libcocotbvhpi_aldec.dll'
ELAB2: Create instances complete.
ELAB2: Elaboration final pass complete - time: 0.1 [s].
KERNEL: Kernel process initialization done.
Allocation: Simulator allocated 6990 kB (elbread=1023 elab2=5814 kernel=152 sdf=0)
KERNEL: ASDB file was created in location C:\temp\cocotb32\examples\endian_swapper\tests\sim_build\wave.asdb
run -all
     -.--ns INFO     cocotb.gpi                         ..mbed\gpi_embed.cpp:71   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     cocotb.gpi                         ..\gpi\GpiCommon.cpp:106  in gpi_print_registered_impl       VHPI registered
     0.00ns INFO     cocotb.gpi                                gpi_embed.cpp:332  in embed_sim_init                  Running on Aldec HDL Simulator version 9.03.2744
     0.00ns INFO     cocotb.gpi                                gpi_embed.cpp:333  in embed_sim_init                  Python interpreter initialized and cocotb loaded!
     0.00ns INFO     cocotb                                      __init__.py:145  in _initialise_testbench           Running tests with cocotb v1.4.0.dev0 from c:\temp\cocotb32\cocotb with cwd C:\temp\cocotb32\examples\endian_swapper\tests\sim_build
     0.00ns INFO     cocotb                                      __init__.py:161  in _initialise_testbench           Seeding Python random module with 1586345967
     0.00ns CRITICAL cocotb.regression                         regression.py:153  in initialise                      Failed to import module test_endian_swapper: No module named 'test_endian_swapper'
     0.00ns INFO     cocotb.regression                         regression.py:154  in initialise                      MODULE variable was "test_endian_swapper"
     0.00ns INFO     cocotb.regression                         regression.py:155  in initialise                      Traceback:
  File "c:\temp\cocotb32\cocotb\__init__.py", line 181, in _initialise_testbench  in initialise                      Traceback (most recent call last):Traceback (most recent call last):

    regression_manager.initialise()
                                                                                                                       File "c:\temp\cocotb32\cocotb\regression.py", line 151, in initialise  File "c:\temp\cocotb32\cocotb\regression.py", line 151, in initialise
    module = _my_import(module_name)
                                                                                                                         module = _my_import(module_name)
                                                                                                                       File "c:\temp\cocotb32\cocotb\regression.py", line 68, in _my_import
ModuleNotFoundError: No module named 'test_endian_swapper'                                                               mod = __import__(name)    mod = __import__(name)

                                                                                                                     ModuleNotFoundError: No module named 'test_endian_swapper'

     0.00ns ERROR    cocotb.gpi                                gpi_embed.cpp:419  in embed_sim_init                  cocotb initialization failed - exiting
     0.00ns ERROR    cocotb.scheduler                            __init__.py:204  in _sim_event                      Failing test at simulator request before test run completion: Simulator shutdown prematurely
sys:1: ResourceWarning: unclosed file <_io.TextIOWrapper name=1 mode='w' encoding='cp1252'>
```

eric-wieser, 2020-04-08:
Your https://github.com/cocotb/cocotb/blob/master/cocotb/share/makefiles/simulators/Makefile.aldec file is not updated from master. You need to merge your local changes with our changes.

pstrueb, 2020-04-08:
Yes you were right. I merged the changes and now I get:
```
(py37_32) C:\temp\cocotb32\examples\endian_swapper\tests>make SIM=aldec TOPLEVEL_LANG=vhdl
make results.xml
make[1]: Entering directory '/c/temp/cocotb32/examples/endian_swapper/tests'
set -o pipefail; GPI_EXTRA= TOPLEVEL_LANG=vhdl \
 MODULE=test_endian_swapper TESTCASE= TOPLEVEL=endian_swapper_vhdl COCOTB_SIM=1 /c/Aldec/Active-HDL9.3/BIN/vsimsa  -do sim_build/runsim.tcl | tee sim_build/sim.log

VHDL/Verilog/EDIF/SystemC Simulator build 9.3.2744.sp1.09.4995

(c) 1997-2014 Aldec, Inc. All rights reserved.

License Number 0

VSIMSA: Configuration files: `C:\Aldec\Active-HDL9.3\vlib\library.cfg', `C:\Aldec\Active-HDL9.3\BIN\vsimsa.cfg'
VSIMSA: `work' is now working library.
 Welcome to VSIMSA!
 This message was printed from `startup.do' macro file.

alib sim_build/work
ALIB: Library `work' attached.
work = C:\temp\cocotb32\examples\endian_swapper\tests\sim_build\work\work.lib
set worklib sim_build/work
VSIMSA: Error: Library `sim_build/work' not found.
acom  -dbg C:/temp/cocotb32/examples/endian_swapper/tests/../hdl/endian_swapper.vhdl
File: .\..\hdl\endian_swapper.vhdl
Compile Entity "endian_swapper_vhdl"
Compile Architecture "impl" of Entity "endian_swapper_vhdl"
Compile success 0 Errors 0 Warnings  Analysis time :  0.2 [s]
asim   +access +w -interceptcoutput -O2 -dbg -loadvhpi "C:/temp/cocotb32/cocotb/libs/libcocotbvhpi_aldec:vhpi_startup_routines_bootstrap" endian_swapper_vhdl
VSIMSA: Error: Working library is not set.
sim_build/runsim.tcl(8) Script: Break at line 8.
        puts [read [open sim.log r]]
sim_build/runsim.tcl(2) Error: Unknown command `puts'
/c/temp/cocotb32/cocotb/share/makefiles/simulators/Makefile.aldec:144: recipe for target 'results.xml' failed
make[1]: *** [results.xml] Error 1
make[1]: Leaving directory '/c/temp/cocotb32/examples/endian_swapper/tests'
c:\temp\cocotb32\cocotb\share\makefiles/Makefile.sim:150: recipe for target 'sim' failed
make: *** [sim] Error 2
```
It appears to me that vsimsa has problems setting the working directory outside the current pwd somehow

themperek, 2020-04-08:
Can you try to change at set work to be in PWD? It just creates a bit os "mess" in PWD but ...

pstrueb, 2020-04-08:
> Can you try to change at set work to be in PWD? It just creates a bit os "mess" in PWD but ...

Okay, I changed sim_build/runsim.tcl 
```alib work
set worklib work```
I Guess I would have to change this in Makefile.aldec to apply this to all simulations ?
The output after I changed this now is:
```
(py37_32) C:\temp\cocotb32\examples\endian_swapper\tests>make SIM=aldec TOPLEVEL_LANG=vhdl
make results.xml
make[1]: Entering directory '/c/temp/cocotb32/examples/endian_swapper/tests'
set -o pipefail; GPI_EXTRA= TOPLEVEL_LANG=vhdl \
 MODULE=test_endian_swapper TESTCASE= TOPLEVEL=endian_swapper_vhdl COCOTB_SIM=1 /c/Aldec/Active-HDL9.3/BIN/vsimsa  -do sim_build/runsim.tcl | tee sim_build/sim.log

VHDL/Verilog/EDIF/SystemC Simulator build 9.3.2744.sp1.09.4995

(c) 1997-2014 Aldec, Inc. All rights reserved.

License Number 0

VSIMSA: Configuration files: `C:\Aldec\Active-HDL9.3\vlib\library.cfg', `C:\Aldec\Active-HDL9.3\BIN\vsimsa.cfg'
 Welcome to VSIMSA!
 This message was printed from `startup.do' macro file.

alib work
ALIB: Library `work' attached.
work = C:\temp\cocotb32\examples\endian_swapper\tests\work\work.lib
set worklib work
VSIMSA: `work' is now working library.
acom  -dbg C:/temp/cocotb32/examples/endian_swapper/tests/../hdl/endian_swapper.vhdl
File: .\..\hdl\endian_swapper.vhdl
Compile Entity "endian_swapper_vhdl"
Compile Architecture "impl" of Entity "endian_swapper_vhdl"
Compile success 0 Errors 0 Warnings  Analysis time :  0.1 [s]
asim   +access +w -interceptcoutput -O2 -dbg -loadvhpi "C:/temp/cocotb32/cocotb/libs/libcocotbvhpi_aldec:vhpi_startup_routines_bootstrap" endian_swapper_vhdl
VSIM: Selected architecture `impl' of entity `endian_swapper_vhdl' from library `work'.
ELBREAD: Elaboration process.
ELBREAD: Elaboration time 0.1 [s].
KERNEL: Main thread initiated.
KERNEL: Kernel process initialization phase.
KERNEL: Time resolution set to 1ps.
ELAB2: Elaboration final pass...
ELAB2: Create instances ...
KERNEL: PLI/VHPI kernel's engine initialization done.
PLI: Loading library 'C:\Aldec\Active-HDL9.3\bin\systf.dll'
PLI: Loading library 'C:\Aldec\Active-HDL9.3\bin\fsdbwriter.dll'
VHPI: Loading library 'C:/temp/cocotb32/cocotb/libs/libcocotbvhpi_aldec.dll'
ELAB2: Create instances complete.
ELAB2: Elaboration final pass complete - time: 0.1 [s].
KERNEL: Kernel process initialization done.
Allocation: Simulator allocated 6990 kB (elbread=1023 elab2=5814 kernel=152 sdf=0)
KERNEL: ASDB file was created in location C:\temp\cocotb32\examples\endian_swapper\tests\wave.asdb
run -all
     -.--ns INFO     cocotb.gpi                         ..mbed\gpi_embed.cpp:71   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     cocotb.gpi                         ..\gpi\GpiCommon.cpp:106  in gpi_print_registered_impl       VHPI registered
     0.00ns INFO     cocotb.gpi                                gpi_embed.cpp:332  in embed_sim_init                  Running on Aldec HDL Simulator version 9.03.2744
     0.00ns INFO     cocotb.gpi                                gpi_embed.cpp:333  in embed_sim_init                  Python interpreter initialized and cocotb loaded!
     0.00ns INFO     cocotb                                      __init__.py:145  in _initialise_testbench           Running tests with cocotb v1.4.0.dev0 from c:\temp\cocotb32\cocotb with cwd C:\temp\cocotb32\examples\endian_swapper\tests
     0.00ns INFO     cocotb                                      __init__.py:161  in _initialise_testbench           Seeding Python random module with 1586347960
     0.00ns INFO     cocotb.regression                         regression.py:189  in initialise                      Found test test_endian_swapper.run_test_001
     0.00ns INFO     cocotb.regression                         regression.py:189  in initialise                      Found test test_endian_swapper.run_test_002
     0.00ns INFO     cocotb.regression                         regression.py:189  in initialise                      Found test test_endian_swapper.run_test_003
     0.00ns INFO     cocotb.regression                         regression.py:189  in initialise                      Found test test_endian_swapper.run_test_004
     0.00ns INFO     cocotb.regression                         regression.py:189  in initialise                      Found test test_endian_swapper.run_test_005
     0.00ns INFO     cocotb.regression                         regression.py:189  in initialise                      Found test test_endian_swapper.run_test_006
     0.00ns INFO     cocotb.regression                         regression.py:189  in initialise                      Found test test_endian_swapper.run_test_007
     0.00ns INFO     cocotb.regression                         regression.py:189  in initialise                      Found test test_endian_swapper.run_test_008
     0.00ns INFO     cocotb.regression                         regression.py:189  in initialise                      Found test test_endian_swapper.run_test_009
     0.00ns INFO     cocotb.regression                         regression.py:189  in initialise                      Found test test_endian_swapper.run_test_010
     0.00ns INFO     cocotb.regression                         regression.py:189  in initialise                      Found test test_endian_swapper.run_test_011
[... 46 lines cut ...]
[... cut ...]
VSIM: Simulation has finished. There are no more test vectors to simulate.
endsim
VSIMSA: Warning: Simulation is not initialized.
ERROR: results.xml was not written by the simulation!
/c/temp/cocotb32/cocotb/share/makefiles/simulators/Makefile.aldec:144: recipe for target 'results.xml' failed
make[1]: *** [results.xml] Error 1
make[1]: Leaving directory '/c/temp/cocotb32/examples/endian_swapper/tests'
c:\temp\cocotb32\cocotb\share\makefiles/Makefile.sim:150: recipe for target 'sim' failed
make: *** [sim] Error 2
```

themperek, 2020-04-08:
So this is not bad. Can you some better (less fancy) example like `adder` ?

pstrueb, 2020-04-08:
> So this is not bad. Can you some better (less fancy) example like `adder` ?

I tried the adder and after again manually modifying the .tcl file for setting the work lib this is the outcome:
``` 
(py37_32) C:\temp\cocotb32\examples\adder\tests>make SIM=aldec TOPLEVEL_LANG=vhdl
make results.xml
make[1]: Entering directory '/c/temp/cocotb32/examples/adder/tests'
set -o pipefail; GPI_EXTRA= TOPLEVEL_LANG=vhdl \
 MODULE=test_adder TESTCASE= TOPLEVEL=adder COCOTB_SIM=1 /c/Aldec/Active-HDL9.3/BIN/vsimsa  -do sim_build/runsim.tcl | tee sim_build/sim.log

VHDL/Verilog/EDIF/SystemC Simulator build 9.3.2744.sp1.09.4995

(c) 1997-2014 Aldec, Inc. All rights reserved.

License Number 0

VSIMSA: Configuration files: `C:\Aldec\Active-HDL9.3\vlib\library.cfg', `C:\Aldec\Active-HDL9.3\BIN\vsimsa.cfg'
 Welcome to VSIMSA!
 This message was printed from `startup.do' macro file.

alib work
ALIB: Library `work' attached.
work = C:\temp\cocotb32\examples\adder\tests\work\work.lib
set worklib work
VSIMSA: `work' is now working library.
acom  -dbg C:/temp/cocotb32/examples/adder/tests/../hdl/adder.vhdl
File: .\..\hdl\adder.vhdl
Compile Entity "adder"
Compile Architecture "RTL" of Entity "adder"
Compile success 0 Errors 0 Warnings  Analysis time :  0.1 [s]
asim   +access +w -interceptcoutput -O2 -dbg -loadvhpi "C:/temp/cocotb32/cocotb/libs/libcocotbvhpi_aldec:vhpi_startup_routines_bootstrap" adder
VSIM: Selected architecture `RTL' of entity `adder' from library `work'.
ELBREAD: Elaboration process.
ELBREAD: Elaboration time 0.1 [s].
KERNEL: Main thread initiated.
KERNEL: Kernel process initialization phase.
KERNEL: Time resolution set to 1ps.
ELAB2: Elaboration final pass...
ELAB2: Create instances ...
KERNEL: PLI/VHPI kernel's engine initialization done.
PLI: Loading library 'C:\Aldec\Active-HDL9.3\bin\systf.dll'
PLI: Loading library 'C:\Aldec\Active-HDL9.3\bin\fsdbwriter.dll'
VHPI: Loading library 'C:/temp/cocotb32/cocotb/libs/libcocotbvhpi_aldec.dll'
ELAB2: Create instances complete.
ELAB2: Elaboration final pass complete - time: 0.3 [s].
KERNEL: Kernel process initialization done.
Allocation: Simulator allocated 5826 kB (elbread=1023 elab2=4652 kernel=149 sdf=0)
KERNEL: ASDB file was created in location C:\temp\cocotb32\examples\adder\tests\wave.asdb
run -all
     -.--ns INFO     cocotb.gpi                         ..mbed\gpi_embed.cpp:71   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     cocotb.gpi                         ..\gpi\GpiCommon.cpp:106  in gpi_print_registered_impl       VHPI registered
     0.00ns INFO     cocotb.gpi                                gpi_embed.cpp:332  in embed_sim_init                  Running on Aldec HDL Simulator version 9.03.2744
     0.00ns INFO     cocotb.gpi                                gpi_embed.cpp:333  in embed_sim_init                  Python interpreter initialized and cocotb loaded!
     0.00ns INFO     cocotb                                      __init__.py:145  in _initialise_testbench           Running tests with cocotb v1.4.0.dev0 from c:\temp\cocotb32\cocotb with cwd C:\temp\cocotb32\examples\adder\tests
     0.00ns INFO     cocotb                                      __init__.py:161  in _initialise_testbench           Seeding Python random module with 1586348772
     0.00ns INFO     cocotb.regression                         regression.py:189  in initialise                      Found test test_adder.adder_basic_test
     0.00ns INFO     cocotb.regression                         regression.py:189  in initialise                      Found test test_adder.adder_randomised_test
     0.00ns INFO     cocotb.regression                         regression.py:417  in execute                         Running test 1/2: adder_basic_test
     0.00ns INFO     ...test.adder_basic_test.0x10feafd0       decorators.py:255  in _advance                        Starting test: "adder_basic_test"
                                                                                                                     Description: Test for 5 + 10
     4.00ns ERROR    cocotb.regression                         regression.py:396  in _score_test                     Test Failed: adder_basic_test (result was ValueError)
                                                                                                                     Traceback (most recent call last):
                                                                                                                       File "C:\temp\cocotb32\examples\adder\tests\test_adder.py", line 21, in adder_basic_test
                                                                                                                         if int(dut.X) != adder_model(A, B):
                                                                                                                       File "c:\temp\cocotb32\cocotb\handle.py", line 690, in __int__
                                                                                                                         return int(self.value)
                                                                                                                       File "c:\temp\cocotb32\cocotb\binary.py", line 442, in __int__
                                                                                                                         return self.integer
[... cut ...]
                                                                                                                     *************************************************************************************

    26.00ns INFO     cocotb.regression                         regression.py:234  in tear_down                       Shutting down...
RUNTIME: RUNTIME_0142 finish called.
KERNEL: Time: 26002 ps,  Iteration: 0,  TOP instance.
KERNEL: stopped at time: 26002 ps
VSIM: Simulation has finished. There are no more test vectors to simulate.
endsim
VSIM: No breakpoints set.
make[1]: Leaving directory '/c/temp/cocotb32/examples/adder/tests'
```

themperek, 2020-04-08:
I think I have seen the same with VHDL. If you could try Verilog also.

themperek, 2020-04-08:
In conclusion:

- remove or correct LICENSE settings
- use `pwd` for `work` library

pstrueb, 2020-04-08:
> I think I have seen the same with VHDL. If you could try Verilog also.

Unfortunately I only have the VHDL licence, sorry!

> In conclusion:
> 
> * remove or correct `LICENSE` settings
> * use `pwd` for `work` library
+ add quotation marks to 
```
-loadvhpi "C:/temp/cocotb32/cocotb/libs/libcocotbvhpi_aldec:vhpi_startup_routines_bootstrap"
```
And use 32 bit python for a 32 bit simulator is only part of the story I would say.
how would I to change Makefile.aldec to use the pwd as work library?

And thanks for helping me debug!

themperek, 2020-04-08:
@pstrueb You are welcome to contribute a patch ;-)

pstrueb, 2020-04-08:
> @pstrueb You are welcome to contribute a patch ;-)

Alright, I will have a look into it. Should I also modify the wiki entry in 
https://github.com/cocotb/cocotb/wiki/Tier-2-Setup-Instructions#conda-based-installation-all-os
to have a solution if you run a 32bit simulator and have conda in 64 bit already installed?

themperek, 2020-04-08:
>to have a solution if you run a 32bit simulator and have conda in 64 bit already installed?
This is exactly what is here: https://github.com/cocotb/cocotb/wiki/Tier-2-Setup-Instructions#conda-based-installation-all-os

pstrueb, 2020-04-08:
> > to have a solution if you run a 32bit simulator and have conda in 64 bit already installed?
> 
> This is exactly what is here: https://github.com/cocotb/cocotb/wiki/Tier-2-Setup-Instructions#conda-based-installation-all-os

You are right, now I get it, too!

commit 2ec67c8f (2020-04-12): Active hdl support (#1614)

Add support for the Aldec Active-HDL simulator

It uses a ``runsim.do`` file with a Tcl-like format (without being full Tcl). 

Fixes #1601
closed 2020-04-08 by pstrueb