cocotb/cocotb #1604: Refactor of Core: RegressionManager, Scheduler, decorators

ktbarrett, 2020-04-08:
Spun off from #1499.

`RunningTest` duplicates all the information available on `test`. This duplication is only necessary because of way that tests are used by the regression manager. The test-specific bits in the scheduler follow from the same poor design choice in the regression manager.

The regression manager:
 * gather tests
 * during initialization starts all the tests (calls the coroutines)
 * if the test initialization fails, it uses the `test` object's info for judgement
 * adds non-skipped `RunningTest` objects to a queue
 * runs each test by scheduling the coroutine
 * because the scheduler determines test end, it must call back into the regression manager (currently via a global reference)
 * pulls info from the `RunningTest` for test judgement, because that's all it has at this point

I think it should go something like this:
 * gather tests and add to queue
 * for each test schedule a regression-manager-bound coroutine that will handle test result
   * Don't need global access because bound method
   * have access to `test` object, so we don't need `RunningTest` duplication
 * initialize and deinitialize a regression-manager-local scheduler for each test

This also brings up the case for deprecating `RunningCoroutine`. It's just a `RunningTask` with a few extra interfaces: `__doc__` (silly), `_parent` (perhaps we should be doing this for `RunningTask`?), other two are redundant if we have parent.

This will decrease code size, complexity, reduce number of special objects, shrink the public interface, make scheduler and regression manager first-class objects, and improve post-test scheduler cleanup.

ktbarrett, 2020-04-08:
One more "feature" I'd like to take into account in the refactor is black-boxing test discovery, so we can experiment with a more pytest-like test discovery in the future.

ktbarrett, 2020-04-10:
I have started to test some of the changes I've made and decided to hold off tearing the scheduler down between tests for now, since I think that will require a lot of changes to the triggers to make them handle cleanup more gracefully. Also will hold off on deprecating `RunningCoroutine` and `RunningTest` for now. They can both be follow up PRs.

Right now the changes

ktbarrett, 2022-03-06:
So I found a way forward with the decoupling. It involves removing all references to tests from the scheduler and injecting behavior that decides when to end the test that is defined on the regression manager.

marlonjames, 2022-03-09:
The approach I started was to add callback functions to the `Task` that are called when it finishes. The regression manager would create the test `Task`, add the callback for handling the result, and then add it to the scheduler. All the scheduler needs to know is to call the callbacks.

ktbarrett, 2025-02-17:
Dupe of #791. xref #4280.

closed 2025-02-17 by ktbarrett