cocotb/cocotb #1611: Unexpected behavior with simple dff example

themperek, 2020-04-08:
From: https://github.com/cocotb/cocotb/pull/1529#pullrequestreview-387867775

```python
# test_dff.py

import random
import cocotb
from cocotb.clock import Clock
from cocotb.triggers import FallingEdge

@cocotb.test()
async def test_dff_simple(dut):
    """ Test that d propagates to q """

    clock = Clock(dut.clk, 10, units="us")  # Create a 10us period clock on port clk
    cocotb.fork(clock.start())  # Start the clock

    for i in range(10):
        await RisingEdge(dut.clk)  # Synchronize ourselves with the clock
        val = random.randint(0, 1)
        dut.d <= val  # Assign the random value val to the input port d
        await FallingEdge(dut.clk)
        assert dut.q == val, "output q was incorrect on the {}th cycle".format(i)
```

```
  6000.00ns ERROR    cocotb.regression                         regression.py:396  in _score_test                     Test Failed: test_dff_simple (result was ValueError)
                                                                                                                     Traceback (most recent call last):
                                                                                                                       File "/examples/dff_simple/test_dff.py", line 20, in test_dff_simple
                                                                                                                         assert dut.q == val, "output was incorrect on the {}th cycle".format(i)
                                                                                                                       File "/cocotb/handle.py", line 409, in __eq__
                                                                                                                         return self.value == other
                                                                                                                       File "/cocotb/binary.py", line 439, in __eq__
                                                                                                                         return self.value == other
                                                                                                                       File "/cocotb/binary.py", line 296, in value
                                                                                                                         return self.integer
                                                                                                                       File "/cocotb/binary.py", line 287, in integer
                                                                                                                         return self._convert_from[self.binaryRepresentation](self._str)
                                                                                                                       File "/cocotb/binary.py", line 193, in _convert_from_unsigned
                                                                                                                         return int(resolve(x), 2)
                                                                                                                       File "/cocotb/binary.py", line 45, in resolve
                                                                                                                         raise ValueError("Unable to resolve to binary >%s<" % string)
                                                                                                                     ValueError: Unable to resolve to binary >z<
  6000.00ns ERROR    cocotb.regression                         regression.py:227  in tear_down                       Failed 1 out of 1 tests (0 skipped)
```

ktbarrett, 2020-04-08:
I'm not a fan of `FallingEdge` here, so I rewrote it slightly.

```python
import random
import cocotb
from cocotb.clock import Clock
from cocotb.triggers import RisingEdge, ReadWrite

@cocotb.test()
async def test_dff_simple(dut):
    """ Test that d propagates to q """

    clock = Clock(dut.clk, 10, units="us")  # Create a 10us period clock on port clk
    cocotb.fork(clock.start())  # Start the clock

    await RisingEdge(dut.clk)  # Synchronize ourselves with the clock
    for i in range(10):
        val = random.randint(0, 1)
        dut.d <= val  # Assign the random value val to the input port d
        await RisingEdge(dut.clk)
        # await ReadWrite()  # uncomment this to make it work
        assert dut.q == val, "output q was incorrect on the {}th cycle".format(i)
```

I haven't tried it, but I feel like (based on experience) Questa using the FLI wouldn't need that statement to work.

themperek, 2020-04-10:
I look into this a bit and I am not sure if there is not some fundamental issue of how initial condition is triggering. It happened that event is triggered at `time==0` and to me not consistent with expectations. 

```python
async def custom_clock(dut):
    while True:
        dut.clk <= 0
        await Timer(5, units="us")
        dut.clk <= 1
        await Timer(5, units="us")

@cocotb.test()
async def test_dff_simple(dut):

    cocotb.fork(custom_clock(dut)) 

    await FallingEdge(dut.clk)
    print("FallingEdge", get_sim_time())

```
Will print `FallingEdge 0` (BTW: verilator gives `FallingEdge 11`)

in Verilog:
```systemverilog
module test_dff_simple();

logic d,q, clk;

initial clk = 0;
always #5us clk = ~clk;

dff dut (q,clk,d);

always@(negedge clk) begin
    $display($time, " negedge is current time.");
end

initial begin
    #10us  $finish;
end

endmodule
```
Will display: `negedge 10`

eric-wieser, 2020-04-10:
@ktbarrett, shouldn't that be `ReadOnly()`?

themperek, 2020-04-10:
Same in this case
```systemverilog
module dff (
    output logic clk, q,
    input logic d
);

initial clk = 0;
always #5us clk = ~clk;

always @(posedge clk) begin
    q <= d;
end

endmodule
```

```python
@cocotb.test()
async def test_dff_simple(dut):

    await FallingEdge(dut.clk)
    print("FallingEdge", get_sim_time())
```

themperek, 2020-04-10:
Something potentially interesting:
```systemverilog

module dff (
    output logic clk
);

initial clk = 0;
always #5us clk = ~clk;

always@(negedge clk) begin
    $display("negedge ",$time);
end

always@(posedge clk) begin
   $display("posedge ",$time);
end

initial #12us  $finish;

endmodule
```
gives:
```
posedge                    5
negedge                   10
```

and this (see `#0`):
```systemverilog
module dff (
    output logic clk
);

initial #0 clk = 0;
always #5us clk = ~clk;

always@(negedge clk) begin
    $display("negedge ",$time);
end

always@(posedge clk) begin
   $display("posedge ",$time);
end

initial #12us  $finish;

endmodule
```

gives:
```
negedge                    0
posedge                    5
negedge                   10
```

ktbarrett, 2020-04-10:
@eric-wieser No. Once we loop back we are in `ReadOnly` and writes will fail. The only way to progress into a writable state at that point is to wait for the next clock cycle. That results in us only writing every other clock. Probably fine for an example, but maybe not.

ktbarrett, 2020-04-10:
> Will print `FallingEdge 0` (BTW: verilator gives `FallingEdge 11`)
> Will display: negedge 10

The value transitions from 'U' -> '0'. VHDL considers this a falling edge. Verilog apparently does not. Verilator just seems off... only 10 or 0 make sense to me.

themperek, 2020-04-10:
Need to make this dff_simple example work in Verilog/VHDL the same way and still be simple.

themperek, 2020-04-10:
Unfortunately, looks like in cocotb we trigger FallingEdge and RisingEdge on 'U' to anything.

ktbarrett, 2020-04-10:
@themperek That last statement is correct. In both languages anything that isn't 'H' or '1' that transitions to 'H' or '1' is a rising edge; and the same for falling edge with 'L' or '0'. The issue with `initial clk = 0` is that it doesn't start out 'U', but with the `#0` delay in there, it does.

The simple solution is not use initial statements and leave the clock signal 'U', `clock.start()` will set that to the '1' or '0' for you. Then you can synchronize with the clock before doing the first write. That *should* work the same in every simulator.

ktbarrett, 2020-04-12:
@themperek has done some experimentation with the example using many simulators and found they are not consistent, even when using the same PLI.

For the following example, the Verilog implementation is used, so all simulators are using VPI.

```python
async def custom_clock(dut):
    while True:
        dut.clk <= 0
        await Timer(5, units="us")
        dut.clk <= 1
        await Timer(5, units="us")

@cocotb.test()
async def test_dff_simple(dut):

    cocotb.fork(custom_clock(dut))

    for i in range(10):
        val = random.randint(0, 1)
        dut.d <= val
        await RisingEdge(dut.clk)
        await ReadWrite() # does not work with modelsim or VCS
        #await Timer(1) # works with modelsim 
        assert dut.q == val, "output q was incorrect on the {}th cycle".format(i)
```

In all the simulators tested besides Modelsim or VCS, `RisingEdge` dumps you *before* the HDL resolves, so you must wait for `ReadWrite` or `ReadOnly` to have the HDL resolve the signals so you can read the correct values. Modelsim and VCS fail when waiting on `ReadWrite` after `RisingEdge`.

We should investigate whether Modelsim and VCS infact dump you *in* `ReadWrite` after a `RisingEdge`, or if some other issue is to blame. Assuming that is the case, what should be done about it?
 
 1. Flowing bugs up to simulators means we are beholden to the simulator developers to fix the issue. It also means there will be differences in behavior version to version with the same simulator.
 2. Make `RisingEdge` also implicitly wait for `ReadWrite`. But we have to ensure this will always do the right thing.
 3. Document the aberrant behavior and move on.

themperek, 2020-04-12:
@ktbarrett The error is `ValueError: Unable to resolve to binary >x<` it is in ModelSim Starter Edition version also.

themperek, 2020-04-13:
Maybe interesting is that this works on everything that I can test as far as I can see:

```python
async def custom_clock(dut):
    while True:
        dut.clk <= 0
        await Timer(5, units="us")
        dut.clk <= 1
        await Timer(5, units="us")

@cocotb.test()
async def test_dff_simple(dut):

    cocotb.fork(custom_clock(dut))

    for i in range(10):
        await FallingEdge(dut.clk)
        val = random.randint(0, 1)
        dut.d <= val
        await RisingEdge(dut.clk)
        await ReadOnly()
        assert dut.q == val, "output q was incorrect on the {}th cycle".format(i)
```

themperek, 2020-04-13:
This also works: 
```

async def custom_clock(dut):
    while True:
        dut.clk <= 0
        await Timer(5, units="us")
        dut.clk <= 1
        await Timer(5, units="us")

@cocotb.test()
async def test_dff_simple(dut):

    cocotb.fork(custom_clock(dut))

    prev_val= val = random.randint(0, 1)
    dut.d <= val

    for i in range(10):
        await RisingEdge(dut.clk)
        val = random.randint(0, 1)
        dut.d <= val
        await ReadOnly()
        assert dut.q == prev_val, "output q was incorrect on the {}th cycle".format(i)
        prev_val = val
```

cmarqu, 2020-04-13:
There is also "The Facts and Fallacies of Verilog Event Scheduling: Is the IEEE 1364 Standard
Right or Wrong?": http://citeseerx.ist.psu.edu/viewdoc/download?doi=10.1.1.611.4282&rep=rep1&type=pdf

ktbarrett, 2020-04-13:
@cmarqu Good find. From just the intro I gather that `ReadWrite` has different behavior on different sims. Is there any way we can work around that? Is there another trigger that always puts us after non-blocking statements resolve? It seems like `RisingEdge` will put us before those statements resolve (we should test that on all simulators as well), so `ReadWrite` not having the behavior discussed in that paper seems pointless. 

@themperek That also explains the `x` value even with the `ReadWrite`. It just leaves us at the same time.

themperek, 2020-04-14:
In some cases, we can use `cbAtEndOfSimTime` but it looks as not in all.

cmarqu, 2020-07-27:
See also #592.

garmin-mjames, 2020-12-03:
> In all the simulators tested besides Modelsim or VCS, `RisingEdge` dumps you _before_ the HDL resolves, so you must wait for `ReadWrite` or `ReadOnly` to have the HDL resolve the signals so you can [...]

Part of the issue may stem from the fact that `dut.clk <= 1` in the `custom_clock` forked task is just scheduling the clock edge for `Scheduler._do_writes()` to perform.
`_do_writes()` waits for `ReadWrite` (and is the first task to run), then updates the signal values. Then other tasks waiting on `ReadWrite` are run.
`RisingEdge(dut.clk)` is triggered after returning from the `ReadWrite` callback, before the HDL resolves, since the signal has changed value.

Modelsim and VCS may have issues registering for another `cbReadWriteSynch` callback with 0 delay right after it was just called.