cocotb/cocotb #1617: Getting rid of None in an option of TestFactory

yannickl96, 2020-04-09:
Hello everyone,

I did not find this in the documentation, so maybe you can push me in the right direction:
I want to use TestFactory to supply a list of discrete values to create different tests. To achieve this, I create a function annotated with @cocotb.test **run_test(dut, my_option=None)** which has the general structure of my test, depending on the value of my_option.
After defining this function, I do the following:
```
factory = TestFactory(run_test)
factory.add_option("my_option", list_of_values)
factory.generate_tests()
```
And it almost works as expected. It creates a test for each of my entries. However, the first test that gets created provides the value None to the parameter my_option. Therefore, I have one more test than intended which obviously fails, as I cannot do anything sensible with None in my test. I assume that this is the result of cocotb running all functions annotated with @cocotb.test if I don't supply the TESTCASE environment variable. Using cocotb.coroutine as an annotation resolves my problem, but I highly appreciate the timeout functionality of the cocotb.test annotation. Is there a clean way how to ensure that the default implementation of my function is not used as a test, without the need to hardcode the test names in TESTCASE?

Cheers and thanks in advance!

ktbarrett, 2020-04-09:
Don't decorate `run_test` with `cocotb.test`, instead use `cocotb.coroutine`.

yannickl96, 2020-04-10:
Hello @ktbarrett ,
as I mentioned, doing that resolves my problem, but I lose the timeout functionality of cocotb.test, or is there another fast way to achieve timeouts in the test function? Anyways, thanks for your quick reply!

eric-wieser, 2020-04-10:
You could use https://docs.cocotb.org/en/latest/triggers.html#cocotb.triggers.with_timeout

eric-wieser, 2020-04-10:
@ktbarrett, could you update the docs for `cocotb.test` to point to `triggers.with_timeout` for an explanation of the timeout arguments?

PR #1630 (2020-04-15): Improve timeout docs and link with_timeout in test decorator
From https://github.com/cocotb/cocotb/issues/1617#issuecomment-612061921. Closes #1617.
closed 2020-04-15 by eric-wieser