cocotb/cocotb #1618: Check for `--enable-shared`

themperek, 2020-04-09:
Should check (in `setup.py`?) if Python was compiled with `--enable-shared` and error if this is not the case.

It looks like on Linux (ubuntu, centos, conda) this is consistent and can be checked with `python -c "import sysconfig; print(sysconfig.get_config_vars('LDSHARED'))"`. Mind that `Py_ENABLE_SHARED ` is not consistant and does not work with conda https://github.com/conda/conda-build/issues/2738

On Windows is not consistent (VS compiler?). I would expect on Windows nobody builds its own python binary.

Do not know on osx.

eric-wieser, 2020-04-09:
Another option would be to invoke distutils to build a trial binary linking to libpython, and see if it fails.

themperek, 2020-05-02:
Some implementation idea:

https://github.com/AdaCore/gnatcoll-bindings/blob/b508bcf68b97d5659b1df32ba17203a896beb95e/python/setup.py#L103-L112