cocotb/cocotb #1636: Add documentation for _sub_handles, _id, _full_name ...

cmarqu, 2020-04-11:
Add documentation for ``_sub_handles``, ``_id``, ``_full_name`` etc. since some of them are also useful for users.

eric-wieser, 2020-04-11:
Note that `_id` currently has a weird default behavior that is specific to vhdl, via the `extended=True` argument. Perhaps we should take the chance to change the behavior before documenting it.

cmarqu, 2020-04-11:
> Note that `_id` currently has a weird default behavior that is specific to vhdl, via the `extended=True` argument. Perhaps we should take the chance to change the behavior before documenting it.

I have added a bit of documentation of the current behavior to the docstring in #1639. The weirdness is now mainly in the default of ``extended`` (and the name which would better be ``extended_only``), and the VHDL-specific code (which can be considered a bug), so can  be done in a different PR.

PR #1639 (2020-07-08): Document "private" members in cocotb.handle
These members are starting with an underscore so as to not collide with
regular HDL names, though most are not really private in the usual meaning
but are useful for user code.

Some type hints are added to aid documentation.

Closes #1636
commit 16fd991c (2020-07-08): Document private members in cocotb.handle

These members are starting with an underscore so as to not collide with
regular HDL names, though most are not really private in the usual meaning
but are useful for user code.

Some type hints are added to aid documentation.

Closes #1636
closed 2020-07-08 by imphil