cocotb/cocotb #1640: Test Verilator issue-234 time changes

wsnyder, 2020-04-12:
Verilator has a new issue-234 git branch (or see here https://github.com/verilator/verilator/pull/2253). This implements `timescale according to the IEEE standard and should make the handling compatible with other simulators.  Can you please check if any cocotb changes are needed to support this?  If any Verilator changes are needed feel free to file issues on verilator or pull requests.

Thanks

Attn @wallento.

themperek, 2020-04-14:
@wsnyder I can confirm that something like this `` `timescale 1us/1ns `` in Verilog works now 👍  

It looks like we need to update here https://github.com/cocotb/cocotb/blob/4c2d7771d5934c25fab00b5b839f85ba1bc8cfdf/cocotb/share/makefiles/simulators/Makefile.verilator#L32

garmin-mjames, 2020-12-03:
So the makefile should be changed to use `--timescale $(COCOTB_HDL_TIMEUNIT)/$(COCOTB_HDL_TIMEPRECISION)`?

themperek, 2020-12-03:
Not sure if `DVL_TIME_PRECISION_STR` is needed at this point and if it still supported?

wsnyder, 2020-12-03:
Yes those are old; you should not need VL_TIME_PRECISION_STR/VL_TIME_UNIT_STR, as you are using `timescale or --timescale (which this issue indicates you are).

PR #2257 (2020-12-04): Use Verilator's --timescale option
Closes #1640
commit 0312036e (2020-12-04): Use Verilator's --timescale option (#2257)

Closes #1640
closed 2020-12-04 by cmarqu