cocotb/cocotb #1642: Document deprecation and removal strategy

cmarqu, 2020-04-12:
We should document our deprecation and removal strategy. 

From https://github.com/cocotb/cocotb/pull/1496#pullrequestreview-387405087

"Only deprecate things in minor releases, and remove them in the next major release. (See https://symfony.com/doc/current/setup/upgrade_major.html#make-your-code-deprecation-free for a project that follows this principle"

cmarqu, 2020-04-12:
What we could do is for every ``.. deprecated::`` line, specify the version number where support is going to be removed.
This would require that we do use predictable future version numbers.

themperek, 2020-04-12:
Last supported versions would be easier.

cmarqu, 2020-04-13:
Would it really? We would have to write something like "Last version this is supported in is 1.x.y", since we cannot know how many x'es and y's may still come.

themperek, 2020-04-13:
>Would it really? We would have to write something like "Last version this is supported in is 1.x.y", since we cannot know how many x'es and y's may still come.

This is exactly what I get with some IC place & route tools.

cmarqu, 2020-11-22:
We should have a section like that in the docs of the 1.5 release: https://docs.antora.org/antora/2.3/whats-new/#caution-coming-soon

PR #2090 (2020-12-22): Updates to CONTRIBUTING.md
Address some of the problems in #2057. Closes #2058. Closes #1642. What I think is still missing is a rather large section, that might be better to put into the Internal Documentation on the Wiki, on contributing to each major codebase (skills, tools, "experts"). That section will not be present in this PR.
closed 2020-12-22 by ktbarrett