cocotb/cocotb #1645: Duplicated loop

tsengr0916, 2020-04-12:
I am seeing some weird behavior that a while loop is executed twice at the same time stamp or in the same clock cycle. The following co-routine is forked-off at beginning of sim, but I see the outer  while loop always executes twice at the same clock cycle. At the end of the routine, I print out some sting and break out, but sim will even print out the string twice then break out the loop.  Could this be a race condition bug in cocotb infrastructure? 

    @cocotb.coroutine
    def _handle_write_data(self):
        while True:
            while True:
                self._wdata_check_done.clear()
                if self.bus.WVALID.bool:
                    WREADY = next(self.wready_generator)
                else: # WVALID == 0
                    if self.bus.WREADY.bool == False:
                        WREADY = next(self.wready_generator)
                    else:
                        WREADY = 1 
                
                self.bus.WREADY.value = WREADY  
                if self.bus.WVALID.value and self.bus.WREADY.value:
                    #yield self.clock.rising_edge()
                    self._wdata_check_done.set(1)
                    break                    
                yield self.clock.rising_edge()

            # Here we sample the information on the write data bus to store into the data fifo            
            wstrb=BitStream(uint=self.bus.WSTRB.uint, length=64).bin
            word=0
            wstrb_bytes=0
            sampled_value=self.bus.WDATA.uint
            shift = 0
            while len(wstrb)>0 and wstrb[-1]=='1' :  
                lsb = sampled_value & 0xFF
                word = (lsb << shift*8) + word
                shift += 1
                wstrb = wstrb[:-1]
                sampled_value = sampled_value >> 8
                wstrb_bytes += 1
                 
            word=hex(word).replace('0x','')
            if len(word)%2 == 1 :
                word = '0' + word
            self.log.info('Final word is {}, iter={}'.format(word, iter))
            wr_byte_array = bytes.fromhex(word)
            wr_byte_array = array.array('B', wr_byte_array)
            wr_byte_array.reverse()
            self._wdata_fifo.append( wr_byte_array )
            self._wstrb_bytes_fifo.append(wstrb_bytes)
            self.log.info("Stored {} to fifo".format(wr_byte_array))
            self._wdata_check_done.set(1)
            yield self.clock.rising_edge()
            print("==> execute !") # this will print out twice 
            break
    
Any ideas?
Thanks
Richard

eric-wieser, 2020-04-12:
It's impossible for that print to appear twice unless you start the coroutine twice - the `break` ensures that.

tsengr0916, 2020-04-13:
Yes Eric. It was a bug in my design. Never mind. Thanks very much!
Richard

closed 2020-04-13 by tsengr0916