cocotb/cocotb #1648: Reword message about the Python interpreter being used

cmarqu, 2020-04-13:
"Did not detect Python virtual environment. Using system-wide Python interpreter" could be read as a problem report by new users. We should reword it.

cmarqu, 2020-04-13:
I would be inclined to just make this a ``LOG_DEBUG`` instead: https://github.com/cocotb/cocotb/blob/e5f888af5a64edc5753a9de2d24e1724d547f5f4/cocotb/share/lib/embed/gpi_embed.cpp#L71

In case we *are* in a virtual environment (and the user likely cares which), its location is reported as an info:
https://github.com/cocotb/cocotb/blob/e5f888af5a64edc5753a9de2d24e1724d547f5f4/cocotb/share/lib/embed/gpi_embed.cpp#L94

eric-wieser, 2020-04-13:
Do we have a means to log the location of the system intepreter? Frankly I'd rather just see in the logs `Using the python interpreter from %s`. It's perhaps still worth doing this from within the C++ code, because it's useful information if the cocotb import fails.

garmin-mjames, 2020-04-22:
We also don't detect if we are in a Conda environment. That may be fine, but logging "Using system-wide Python interpreter" in that context may be confusing.

themperek, 2020-04-22:
We could chek fo and display `PYTHONHOME` variable (if exist).

ktbarrett, 2025-10-26:
This has been changed to simply display the Python environment being used in 2.0.