cocotb/cocotb #1650: Recently cocotb and ghdl have stopped working together.

benreynwar, 2020-04-13:
The error is:
``
/usr/local/bin/ghdl: symbol lookup error: /home/travis/build/benreynwar/check_cocotb_ghdl/cocotb/cocotb/libs/libcocotbvpi_ghdl.so: undefined symbol: vpi_register_systf
``
I've created a dummy travis run to illustrate this:

https://travis-ci.org/github/benreynwar/check_cocotb_ghdl

I've also opened an issue in ghdl (https://github.com/ghdl/ghdl/issues/1222).

eric-wieser, 2020-04-13:
Mind taking a look @themperek?

themperek, 2020-04-13:
It looks ok here: https://travis-ci.org/github/cocotb/cocotb/jobs/674472863#L1126 (latest master)

benreynwar, 2020-04-13:
It looks like this is a ghdl issue and was introduced in the last couple of days.  In the ghdl issue tgingold said he thinks he knows what the issue is, so I don't think cocotb needs to worry about it.

themperek, 2020-04-13:
@benreynwar Thank you for update.

PS. Still looking for someone who can clean up our GHDL tests in cocotb.

closed 2020-04-13 by themperek

benreynwar, 2020-04-13:
tgingold has fixed the issue now.  He had a comment about fixing a cocotb makefile.  I don't think any changes are necessary, but it would be good if someone who better understands what's happening takes a look:

See ghdl issue:
https://github.com/ghdl/ghdl/issues/1222