cocotb/cocotb #1655: Document simulator timing behavior

ktbarrett, 2020-04-14:
Adds some documentation regarding common questions that some people may ask about the timing triggers `ReadWrite` and `ReadOnly`, what state are we in when an edge trigger (e.g. `RisingEdge`) or `Timer` occurs, and how to use these primitives effectively for common problems like modelling combinational and registered logic.

sfavazza-duagon, 2020-07-07:
From what I have understood so far:

- **NextTimeStep**: this routine consumes simulation time till the beginning of the next time step, i.e. the simulation instant just before the delta cycles are processed.
- **ReadOnly**: this routines consumes simulation time until the end of the current time step, i.e. the phase where all signals get their values updated and no more changes are allowed.
-  **ReadWrite**: this is a real mystery for me...
- **Rising/FallingEdge**: despite it seems clear from its own name, I do not understand at which of the aforementioned time phases the simulator points.

Perhaps a clarification in this thread would be appreciated, then I would be glad to properly document it and ask for a merge request.

ktbarrett, 2020-07-07:
## Explanation
From my understanding, these steps *in order*. A diagram might be useful for the documentation.

1. Value Change: these triggers leave you at the instant the signal changes to the aforementioned value, no dependent signals have had a chance to resolve. Note that this may put you in the middle of a time step depending on who is driving this signal. 
2. `ReadWrite`: this places you after all the combinationally-dependent HDL signals have resolved, but you can still write. This allows you to read the combinational outcome of an event and react. All cocotb writes are executed at the very end of this stage.
3. `ReadOnly`: this places you after all cocotb values have been written and combination-dependent HDL signals have resolved. You aren't allowed to write in this stage.
4. `NextTimeStep`: as you said, this places you at the beginning of the next set of changes, whatever that may be.

## Caveats
Because of how these work, you cannot wait for the `ReadWrite` stage in `ReadOnly`. Any other serial combination is valid. Some simulators throw errors if you attempt to do this (like Icarus), others (like Questa and Riviera) just wait until the next time step's `ReadWrite` phase. This is #1245.

The implementation of the VPI/VHPI/FLI triggers is all a bit different simulator to simulator and so while the above is the ideal behavior, it's not always the behavior you will see. This is #1611 and all the related issues.

## Remaining Questions
* How does `ReadWrite` and `ReadOnly` work in combinational statements *with delays*, like are present in back-annotated sims?
* `ReadWrite` or `ReadOnly` after a `Timer` trigger is seemingly nonsense, but I'm guessing it just works with whatever the currently executing time step is, or skips to the next one?

garmin-mjames, 2020-09-08:
> ReadWrite: this places you after all the combinationally-dependent HDL signals have resolved, but you can still write. This allows you to read the combinational outcome of an event and react. **All cocotb writes are executed at the very end of this stage.**

The saved/delayed/cached writes actually happen at the beginning of this stage, so that any coroutines waiting on `ReadWrite()` will see the signal values from the writes.

> `ReadWrite` or `ReadOnly` after a `Timer` trigger is seemingly nonsense, but I'm guessing it just works with whatever the currently executing time step is, or skips to the next one?

I think they will just register the callback for the current time step. All `Timer` callbacks happen at the beginning of the time step.

ktbarrett, 2022-05-10:
Documentation added to Wiki. https://github.com/cocotb/cocotb/wiki/Timing-Model

closed 2022-05-10 by ktbarrett