cocotb/cocotb #1657: Add type checking to handle non-blocking writes

ktbarrett, 2020-04-16:
When using `setimeddiatevalue` to set a signal's value, it does type checking to ensure it can pass a valid value to the C layer. However, the non-blocking write methods `value` and the syntax `<=` depend upon the type checking in `setimmediatevalue`. If these non-blocking writes are scheduled, the type checking it not done immediately. The exception will throw in the internal writer coroutine instead of at the use site.

Given this code:

```python
@cocotb.test()
async def test_breaks(dut):
    dut.stream_in_data <= "wrong type"
    await cocotb.triggers.Timer(10)
```

I get this stacktrace:
```
Traceback (most recent call last):
  File "/home/kaleb/dev/cocotb/cocotb/scheduler.py", line 260, in _do_writes
    handle.setimmediatevalue(value)
  File "/home/kaleb/dev/cocotb/cocotb/handle.py", line 648, in setimmediatevalue
    .format(type(value), value))
TypeError: Unsupported type for value assignment: <class 'str'> ('wrong type')
```

Move type/value checking for a given `SimHandle` subtype into a central function and use it in all write methods.

eric-wieser, 2020-04-16:
My fear is it's worse than that, and all future writes fail because the writing thread crashed

ktbarrett, 2020-04-16:
In the writer coroutine I'm guessing we should catch, log, and squash?

eric-wieser, 2020-04-16:
I have a better plan, which I started prototyping but don't have time today to finish. The idea would be to queue the fully-processed simulator calls from `handle.py`, with no further checks or computations. I'll push it to a branch for you to take a look.

closed 2020-04-16 by eric-wieser