cocotb/cocotb #1660: Deemphasize/remove endian swapper example

imphil, 2020-04-16:
The endian swapper example is a nice advanced use case, but it's also an odd one:
- depends on swig
- requires user to understand how to write a custom Python extension module
- Ultimately does something rather boring.

Still we list is rather prominently in our docs and README. 

I'd go for either 
1. fully remove the example until we come up with something better,
2. or: de-emphasize it. Keep it working, but don't list it as first tutorial on how to use cocotb (it's more of a "btw if you want to go fancy, that's what you could do")

themperek, 2020-04-16:
I propose this as first example https://github.com/cocotb/cocotb/pull/1620
I was thinking to make a short tutorial on this while changing: (change clock edge, add coroutine ... )

eric-wieser, 2020-04-16:
I'd vote option 2, but also do a better job of framing what it is trying to show: a case where you have an existing C hardware driver, and want to run it against hardware instead.

As an aside: the way we build the two C extension modules in that example, `io_module` and `hal`, and link them together, is explicitly unsupported by python, and not guarantee to work on all platforms.

imphil, 2020-04-16:
@eric-wieser if you'd have to write such code, would you really go for the implementation approach that was chosen here? I have had a fantastic experience with cython, that's what I'd go for for such an example.

eric-wieser, 2020-04-16:
I've not spent much time around cython. Since the example is so minimal, for this case I might just:

* Replace the link-time injection of `IORD` with runtime injection - ie, make `endian_swapper_init` take an object like:
  ```
  {
      int (*read_func)(void* self, int addr);
      int (*write_func)(void* self, int addr, int value);
  }
  ```
  This eliminates the need for the `io_module` extension module. It's also representative of what some production driver code I wrote in the past looked like.
* Use `ctypes` to construct this table of function pointers, and call the few functions we care about.

A cython example would likely be valuable too.

ktbarrett, 2025-10-26:
This was removed at some point in the past (1.5?).

closed 2025-10-26 by ktbarrett