cocotb/cocotb #1662: Change VPI callback for ReadWrite

ktbarrett, 2020-04-16:
Discussion of proposed solution following #1611. #1060 likely related.

### Background
After waiting on an edge trigger on a signal, we exist in a state where computations based on the change in value on the signal have not been resolved. Reading values at this point yields previous values for the related signals. If want to read the resolved values and respond in the same time cycle (modelling combinational logic), we need to move to a state where the HDL assignments are resolved, but we can still write values. This is the point of `ReadWrite`.

We currently use the `cbReadWriteSynch` trigger to move us into that context; however, as https://github.com/cocotb/cocotb/issues/1611#issuecomment-613076767 points out, there is some contention on what that trigger actually means. @themperek has done investigative testing showing that Questa and VCS's implementation of `cbReadWriteSynch` *do not* put us after the values resolve. So the `ReadWrite` trigger behaves differently under different simulators. There is no *proper* way to model combinational logic in cocotb in those simulators (you could still use `NextTimeStep` or `Timer(1)`, but the timing will be off).

### Solution
If we move the implementation of `ReadWrite` to use the `cbAtEndOfSimTime` trigger, VCS will function as intended. However, Questa's VPI implementation is rather poor and does not implement this trigger. So with Questa we can choose between an implementation of `cbReadWriteSynch` that doesn't work how we want it to, or something that doesn't work at all. We should leave Questa alone for right now, the regressions pass and existing code (likely) works and will continue to work.

Kudos to @themperek for doing all the investigative legwork on this.

imphil, 2020-04-16:
@mballance do you happen to know someone who can look at the Questa side of things here, and let us know if there is another solution we haven't seen so far?

mballance, 2020-04-16:
@imphil, good question. I'll be happy to take a look at it myself initially, though it won't be until the weekend. Is there a testcase that I can use to observe the current behavior?

themperek, 2020-04-16:
Maybe on simple D flip-flop ?

```python
import random
import cocotb
from cocotb.clock import Clock
from cocotb.triggers import RisingEdge

async def custom_clock(dut):
    while True:
        dut.clk <= 0
        await Timer(5, units="us")
        dut.clk <= 1
        await Timer(5, units="us")

@cocotb.test()
async def test_dff_simple(dut):

    cocotb.fork(custom_clock(dut))

    for i in range(10):
        val = random.randint(0, 1)
        dut.d <= val
        await RisingEdge(dut.clk)
        await ReadWrite() # fails to sample correct dut.q
        #await Timer(1) # works
        assert dut.q == val, "output q was incorrect on the {}th cycle".format(i)
```

mballance, 2020-04-19:
@ktbarrett, I played around a bit with the Python @themperek shared, and the 'dff' DUT from the README. I could use a bit more information from you. When using the ReadWrite trigger, my guess is that users will want to see updated inputs based on the outputs assigned by cocotb. Is this correct? 
Here's the modified DUT that I'm using

```
`timescale 1us/1ns

module dff (
    output logic q,
    input logic clk, d
);

wire d_res = d;

always @(posedge clk) begin
    q <= d;
end

endmodule
```

I'm still driving `d`, but I'm observing `d_res` instead of `d` directly. What I'm observing with Questa is that I see `d_res` == `val` as long as we use  vpiNoDelay for putting values (https://github.com/cocotb/cocotb/blob/1a4d7bf5a3251afd8cde5e52d114d15c497b525c/cocotb/share/lib/vpi/VpiCbHdl.cpp#L386). 
With this approach, I'm still not able to see the updated value of net being driven (`d`). 

How close does this get us to what is needed for modeling combinational logic?

ktbarrett, 2020-04-19:
I did some research on why we are using `vpiInertialDelay` and why the two are different and there is quite the history, #1014, #768 and their related PRs. According to https://github.com/cocotb/cocotb/issues/1014#issuecomment-511113211, using `vpiNoDelay` causes issues. This is likely *one* of the problems.

I don't think the example highlights what I mean though, I will come up with an example and test to demonstrate what I mean.

ktbarrett, 2020-04-19:
@mballance What I am talking about is *reacting* combinationally. For example, you have FIFO that a `valid` output, and you need cocotb to read on that valid output and in reaction set an acknowledgement signal.

Simply, this is;
```python
while True:
    await RisingEdge(dut.clk)
    await ReadWrite()
    if str(dut.rdValid.value) == '1':
        dut.rdEn.value = '1'
        _ = dut.rdData.value  # read data
```

If we don't put the `ReadWrite` there, we are reading the previous clock cycle's value. This still works by accident though...

marlonjames, 2021-03-13:
Aldec doesn't support `cbAtEndOfSimTime`.

It should also be noted that some simulators may interpret the _cb_data_p->time_ field differently. Icarus interprets it as an absolute time for `cbAtStartOfSimTime` and `cbAtEndOfSimTime`, but Aldec sims interpret it as a delay for the supported `cbAtStartOfSimTime`, like for read-write/read-only.

eric-wieser, 2021-04-26:
ON a related note - should `signal <= value `assignments be executed at `cbNBASynch` instead of the `cbReadWriteSynch` where they are currently executed?

ktbarrett, 2021-04-26:
> ON a related note - should `signal <= value `assignments be executed at `cbNBASynch` instead of the `cbReadWriteSynch` where they are currently executed?

I think so. Supporting both would be a good idea, and doing writes during `cbNBASynch` so that cocotb writes are treated no differently than regular non-blocking assignments would be nice. The biggest issue still being horrible simulator support.

eric-wieser, 2021-05-11:
Do we know if Questa supports `cbNBASynch`? It doesn't look like we tested that above.

Donmar001, 2023-12-14:
@cmarqu @eric-wieser @mballance @imphil @themperek Asking help,How to identify the DFF in the code by the VPI,use the vpi_get_str to map the name?