cocotb/cocotb #1671: Struct support at top module interface

chiraag, 2020-04-19:
Are structs supported at the interface of the top-level system verilog module? If yes what is the appropriate syntax to add them to the signal list in the driver? Should we pack them manually in Python and then assign them to the appropriate port?

elgorwi, 2020-04-23:
@chiraag, some more details of which simulator and a code snippet would help us understand the problem better. 

Following code seems to indicate that a ctypes structure could be used to drive a ModifiableObject:
https://github.com/cocotb/cocotb/blob/dfaac9732e3b6b07e37c928c25bc93b7cc9e6d77/cocotb/handle.py#L647

However, it is likely that the verilog struct would be a HierarchyObject rather than a ModifiableObject. Have you tried the dot syntax to address the individual fields of the struct?

chiraag, 2020-04-24:
@elgorwi I have not tried the dot syntax yet but I will give a shot today. If that does not work I will post a small reproducible test-case

closed 2020-11-24 by github-actions[bot]