cocotb/cocotb #1675: Test abortion methods don't work

ktbarrett, 2020-04-19:
`RunningTest` objects have `abort` methods, the scheduler has the `finish_test` method. I have run several tests and found that neither actual ends the test with the correct outcome. `finish_scheduler` does work, but doesn't clean up before scheduling the next test.

Obviously no one is using these methods or we would have complaints. As a part of the scheduler refactor I vote we remove all of these methods and put a test abortion method in the regression manager (which will be the only things holding a reference to the current test after the refactor).

eric-wieser, 2020-04-19:
Note that users have no sane way to obtain a `RunningTest` instance in the first place.

ktbarrett, 2020-04-19:
Note these functions are used in `_sim_event` and work in that context, but don't work in user code. This probably has to do with sim event not going through the simulator. More tests to follow.

garmin-mjames, 2020-04-28:
`finish_test` and `finish_scheduler` are designed to work when the simulator has control and no coroutines are scheduled, since they come up through `gpi_embed_event`.

ktbarrett, 2020-04-28:
They might have originally been intended that way, but we need a way for the user to finish the currently running test with an outcome so that we can one day move away from `TestComplete`.

eric-wieser, 2020-04-28:
Perhaps that way could simply be a wrapper around a test that waits for `First(test_coro, abort_test_event)`

ktbarrett, 2025-02-21:
These function were made private in 2.0.

Test should be ended with the following methods:
```python
raise TestSuccess  # cocotb 1.X
cocotb.pass_test()  # cocotb 2.X
assert cond  # where cond is False
raise Exception
```

closed 2025-02-21 by ktbarrett