cocotb/cocotb #1676: Document performance improvements suggestions

themperek, 2020-04-19:
Add a documentation section that can be done for improving performance.

Some initial suggestions can be (please add more):

- Check optimization options of your simulator (default from cocotb may not be optimal for performance).
- Use `async`/`await` syntax.
- Try to minimize the number of triggers (`await`/yield`):
   - Generate clocks on the HDL side
   - Used slowest possible clock/(or use signal/strobe as a trigger) to exchange data between python and HDL
   - Exchange a larger amount of data on one trigger (parallel) than multiple triggers.
- Use profiler (`COCOTB_ENABLE_PROFILING=1 make ... `)

cmarqu, 2020-04-19:
* Reduce the amount of log messages

nndurj, 2020-04-30:
Is it possible to import verilated design as ctypes module to avoid interprocess communication? See 
https://github.com/myhdl/myhdl/pull/338. 
>No interprocess communication is used since the verilated design is imported as a ctypes module.

ktbarrett, 2020-05-04:
Instead of reducing the amount of log messages, we could use a [faster log implementation](https://github.com/brmmm3/fastlogging).

jshaker000, 2020-12-28:
I think with Xcelium / Questa, a big part of the performance hit is that we have to compile / elaborate the design access +rwc (because cocotb can in theory read / write to ANY net in the whole design). And so the compilers can make less optimizations.   
   
Unfortunately, I dont see a very good way around this. As a test I've run plain SV testbenches both with and without rwc and I see a 2-2.5x speed difference typically.  
   
I guess one solution would be specify in compiling which nets you actually want to read / write specifically and specify only those in access, but then that starts to make the cocotb flow much less appealing to maintain. Its tough ..

cmarqu, 2020-12-28:
I suppose we could have a "learning mode" where we record every path that is accessed and use this information in a second run - VCS even has such a mode built in.
It would not necessarily work so well with randomization since different, previously unrecorded, paths could be taken.

themperek, 2020-12-28:
It may need to part of the design to actually know what one is doing. We may help by documenting this and giving (easy) way to change.