cocotb/cocotb #1680: Adding cocotbext-wishbone as cocotb group

Martoni, 2020-04-20:
[cocotbext-wishbone](https://github.com/wallento/cocotbext-wishbone) module is not enough known as we've seen in this[ issue message](https://github.com/cocotb/cocotb/pull/439#issuecomment-615367546).

It could be great to include this project/module in «[cocotb github organization](https://github.com/cocotb)» no ?
As explain in the[ cocotb documentation.](https://github.com/cocotb/cocotb/blob/master/documentation/source/extensions.rst#code-hosting)

eric-wieser, 2020-04-20:
Cleaning up https://github.com/cocotb/cocotb/wiki/Further-Resources may also be valuable here. There are a lot of links without much description there.

Martoni, 2020-04-20:
I added a section in [Further-Resources](https://github.com/cocotb/cocotb/wiki/Further-Resources#extension-modules-cocotbext)

themperek, 2020-05-07:
@Martoni Please see here: https://docs.cocotb.org/en/latest/extensions.html#code-hosting

Martoni, 2020-05-11:
Ok, as I understand, I have to add some unit tests and include the extension do PyPi.
And make a sphinx documentation instead of simple github README.

cmarqu, 2020-05-11:
Documentation in Sphinx is not a must in my opinion, but it has the advantage that you can automatically reference cocotb documentation via https://www.sphinx-doc.org/en/master/usage/extensions/intersphinx.html