cocotb/cocotb #1681: Plan for xsim and/or quartus simulator support in cocotb?

DavideConficconi, 2020-04-20:
Hello,
I am a newbie to cocotb so maybe is not a real issue but I was not able to found an answer to my doubts here and in the read-the-docs.

Is there any plan to support, or it is already supported -- in case sorry for this issue --, Xilinx and Intel simulators with cocotb?

Thanks!

themperek, 2020-04-20:
As I am aware `xsim` does not support any Procedural Interface (VPI/VHPI) so this is not possible. As I know Intel uses to Modelsim that in principle should work (is working in older versions).

If this does not answer your question feel free to re-open.

closed 2020-04-20 by themperek

xobs, 2020-06-24:
Adding some more information to convince myself why cocotb still doesn't have support for xsim:

I believe the only current programmatic interface available within xsim is "DPI", which (and correct me if I'm wrong here) doesn't support making callbacks on signal events, which is crucial to how cocotb operates. "VPI" has, apparently, been on the roadmap for years but hasn't made it in because apparently their customers aren't demanding it.

There is a thing called "IKI" that looks like it could be used, but it is not documented at all.