cocotb/cocotb #1682: Wait on event and condition in `c/c++`.

themperek, 2020-04-20:
For performance reasons, it would be probably useful to have the ability to wait on condition in `c/c++`.  Looking at monitors/drivers that could have a performance gain if used correctly.

Something like below to wait in `c` and not in `python`:
```python
while (condition ==x):
    await event
```

Example python syntes can be like in #1250:
```python
await RisingEdge(dut.clock).until_after(lambda: dut.ack.value)
```

Can this be a new callback in GPI or some better ideas?

I think @eric-wieser @ktbarrett has some ideas.

eric-wieser, 2020-04-20:
Right now it is impossible to have more than one C-level callback for a signal edge. The issue is that we allocate the callbacks statically per signal:

https://github.com/cocotb/cocotb/blob/c4cfcd3e3faae6ea12851c0336fa92924ffce176/cocotb/share/lib/vpi/VpiImpl.h#L188-L190

I imagine this was motivated by a reluctance to implement proper memory management of these callbacks. Allocating on the heap would be a bad idea unless we were capable of cleaning up after ourselves.

themperek, 2020-04-20:
But doing like bellow can get the same (with some limitations) :
```python
await RisingEdge(dut.signal)
await RisingEdge(dut.clk)
```

eric-wieser, 2020-04-20:
That's not correct, it gives a false positive if a pulse appears on `signal` between clock edges. You need
```python
while True:
    await RisingEdge(dut.clk)
    if dut.signal.value.binstr == "1":
        break
    await Edge(dut.signal)
```

themperek, 2020-04-20:
@eric-wieser But this is not implemented like this: https://github.com/cocotb/cocotb/blob/b5780f3f478ddaff4629cf46e8c998ebebc3b9ad/cocotb/triggers.py#L910-L921 ?

eric-wieser, 2020-04-20:
That code isn't specific to edges, but to any complex condition. I'm also starting to suspect that `ReadOnly` is wrong, and that the signal should be captured immediately at the rising edge, to match how the verilog would capture it.

eric-wieser, 2020-04-20:
I'm also starting to suspect that `ReadOnly` is wrong, and that the signal should be captured immediately at the rising edge of `clock`, to match how the verilog would capture it.

themperek, 2020-04-20:
BTW: For `ClockCycles` one can use `Timer` in simple cases.

ktbarrett, 2020-04-20:
@eric-wieser

After looking at it, it doesn't seem like it would be a herculean task to add support for multiple waiters on a signal change (at least to VPI, not sure about other interfaces). Make the callbacks vectors, each callback holds a reference to the signal object that it can use that to remove itself from the vector once it cleans up. They just have to be friends (or have a deregister method). The `value_change_cb` method would return a new callback object per call.

closed 2020-11-17 by themperek

ktbarrett, 2020-11-17:
@themperek Does this warrant closing? Or are we counting on #1689 to capture our thoughts?

themperek, 2020-11-17:
I think #1689 and #2208 are enough.