cocotb/cocotb #1687: Document how cocotb can be bolted onto an existing simulation environment (bypassing makefiles...)

gretzteam, 2020-04-22:
Adding cocotb support to an existing simulation environment is doable without the use of Makefiles. Document what needs to be done for each simulator.

Snippet from Gitter discussion:

> At this point, bolting in cocotb into an existing environment is really just a case of adding -m libcocotbvpi_icarus or similar to the simulator invocation, and setting the environment variables MODULE, TOPLEVEL, TOPLEVEL_LANG, and COCOTB_SIM. .. For other simulators, the flag for libcocotb* is slightly different

>xcelium: -loadvpi /cocotb/cocotb/libs/libcocotbvpi_ius:vlog_startup_routines_bootstrap

>We may want to rename MODULE to COCOTB_MODULE etc. so that it's clear what this is about when used to together with other envvars from the original testbench environment

themperek, 2020-04-22:
Related to #1461 #1562

garmin-mjames, 2020-05-07:
See #1718

cmarqu, 2022-02-28:
Closed a while ago by https://github.com/cocotb/cocotb/pull/2340.

closed 2022-02-28 by cmarqu