cocotb/cocotb #1688: Incomplete cleanup in scheduler leading to bugs

ktbarrett, 2020-04-22:
The test that caused the error was added in ktbarrett/cocotb@d589303f9ae8cca17789b21b597e4bec66cf80bf. The failure can be observed by running the failing test and at least one other test after it.

```
cd tests/test_cases/test_cocotb
make MODULE=test_concurrency_primitives COCOTB_SCHEDULER_DEBUG=1 TESTCASE="test_exceptions_forked_late,test_exceptions_first"
``` 
[sim.log](https://github.com/cocotb/cocotb/files/4516438/sim.log)

You can see from the log that the coroutine `test_exceptions_forked_late.<locals>.await_later` that was scheduled in the first test was not cleaned up and fires in the second test and causes it to fail.

ktbarrett, 2020-04-22:
The test was originally written for #1526, with the current scheduler semantics the test should fail with `ValueError` and the next test should pass.

ktbarrett, 2020-04-22:
The issue seems to be that when `await_later` is `schedule`d it chains up to the point of failure in the forked coroutine which calls `cleanup` and then `_test_complete` which resets the `_terminate` flag allowing it to skip over the case that is supposed to handle that.

The real answer is that the scheduler shouldn't be failing tests in `unschedule`, that checking for test completion or performing cleanup should be handled *outside* of the scheduler, and that the scheduler shouldn't start the next test from inside the scheduler!

I'll try to find an easy way around this...

eric-wieser, 2020-04-24:
So to be clear, the failure mode is:
```
     0.00ns INFO     cocotb.regression                         regression.py:194  in initialise                      Found test test_concurrency_primitives.test_exceptions_forked_late
     0.00ns INFO     cocotb.regression                         regression.py:194  in initialise                      Found test test_concurrency_primitives.test_exceptions_first
     0.00ns INFO     cocotb.regression                         regression.py:422  in execute                         Running test 1/2: test_exceptions_forked_late
     0.00ns INFO     ..ptions_forked_late.0x7f31faab5be0       decorators.py:253  in _advance                        Starting test: "test_exceptions_forked_late"
                                                                                                                     Description: Test exception propogation via fork, but await *after* forked coro has raised
     0.00ns ERROR    ..locals>.raise_soon.0x7f31faad8160        scheduler.py:502  in unschedule                      Exception raised by this forked coroutine
     0.00ns ERROR    cocotb.regression                         regression.py:401  in _score_test                     Test Failed: test_exceptions_forked_late (result was ValueError)
                                                                                                                     Traceback (most recent call last):
                                                                                                                       File "/home/kaleb/dev/cocotb/tests/test_cases/test_cocotb/test_concurrency_primitives.py", line 107, in raise_soon
                                                                                                                         raise ValueError('It is soon now')
                                                                                                                     ValueError: It is soon now
     0.00ns INFO     cocotb.regression                         regression.py:422  in execute                         Running test 2/2: test_exceptions_first
     0.00ns INFO     ..t_exceptions_first.0x7f31faaccda0       decorators.py:253  in _advance                        Starting test: "test_exceptions_first"
                                                                                                                     Description:  Test exception propagation via cocotb.triggers.First 
     0.00ns ERROR    ..ocals>.await_later.0x7f31faad81d0        scheduler.py:502  in unschedule                      Exception raised by this forked coroutine
     0.00ns ERROR    cocotb.regression                         regression.py:401  in _score_test                     Test Failed: test_exceptions_first (result was ValueError)
                                                                                                                     Traceback (most recent call last):
                                                                                                                       File "/home/kaleb/dev/cocotb/tests/test_cases/test_cocotb/test_concurrency_primitives.py", line 113, in await_later
                                                                                                                         await future
                                                                                                                       File "/home/kaleb/dev/cocotb/cocotb/decorators.py", line 200, in __await__
                                                                                                                         return (yield self)
                                                                                                                       File "/home/kaleb/dev/cocotb/tests/test_cases/test_cocotb/test_concurrency_primitives.py", line 107, in raise_soon
                                                                                                                         raise ValueError('It is soon now')
                                                                                                                     ValueError: It is soon now
     0.00ns ERROR    cocotb.regression                         regression.py:232  in tear_down                       Failed 2 out of 2 tests (0 skipped)
```

ktbarrett, 2020-04-24:
@eric-wieser that is correct. Notice the coroutine from the first test kills the second test.

eric-wieser, 2020-04-24:
It seems weird than that, the same outcome is being used to end both tests.

ktbarrett, 2020-04-24:
@eric-wieser Did you check the log from the OP? You can see a couroutine from the first test is still being scheduled during the second test (line 59). After the `Timer` returns it reschedules that coroutine where it awaits on the outcome from `raise_soon`, which is stored.

eric-wieser, 2020-04-24:
Thanks, now I understand

garmin-mjames, 2020-04-29:
Copied my comment from #1690:

---

As you stated, this is another issue due to `add` scheduling a new coroutine while the first is in the middle of `schedule`. `cleanup` doesn't clean up everything because it has no way to get a reference to `await_later`.

1. The test forks `_check_traceback` and joins on it through `_trigger_from_unstarted_coro`.
2. `_check_traceback` forks `await_later` and joins on it through `_trigger_from_unstarted_coro` also.
3. `await_later` forks `raise_soon` directly, which means `raise_soon` runs immediately, before `await_later` can `await` a trigger.
4. `raise_soon` raises the exception, `unschedule` calls `abort` on the test, which calls `unschedule`, and `cleanup` kills all forked coroutines that are waiting on a trigger.
5. The stack is unwound back up into `await_later`, and it happily continues on it's way to `await Timer(1)`.
---
Because the `Timer(1)` trigger is added after `cleanup`, it fires in the next test (see line 57).

ktbarrett, 2021-12-09:
The correct solution is to not use `fork`, but prefer `start_soon` instead.

closed 2021-12-09 by ktbarrett

Febbe, 2024-02-27:
This is also an issue with `start_soon`, why is this issue closed?
cocotb          1.8.1

ktbarrett, 2024-02-27:
@Febbe open a new issue with a minimal failing example.