cocotb/cocotb #1689: Conditional waiting on events : API discussion

themperek, 2020-04-22:
Based on https://github.com/cocotb/cocotb/pull/1250 (by @Fatsie) proposals and discussion in https://github.com/cocotb/cocotb/issues/1682

We have discussed a bit internally on https://github.com/cocotb/cocotb/pull/1250 and concluded is a that it is a good idea but:
- function names have to be clear
- we can gain performance if we implement this in a slightly different way 

For performance reasons the function should include as input: 
- synchronization trigger (clock and edge)
- sensitivity list (signals to observe)
- condition for trigger

Implementation of the waiting part should be somewhere along the lines (from @eric-wieser ) :
```python3
while True:
    await RisingEdge(dut.clk)
    if condition():
        break
    await First(*[Edge(hdl) for hdl in signals])
```
We should support signals, vectors and buses? Should we use ReadOnly()?

This way we minimize the number of triggers/callbacks that will improve performance and may become the recommended way to wait for the signal.
In the future, such a trigger can be moved to GPI/C++.

Considering those points we should agree on API before implementation.
Proposals are welcome. https://github.com/cocotb/cocotb/pull/1250 can be a starting point.

ktbarrett, 2020-04-22:
Just to be clear, what we are talking about is creating a function *specifically* for improving the performance of synchronizing ourselves on registered logic, correct? I think in that case we should specifically call out the use case with the word "registered" in the name. I'm thinking `RegisteredEdge`, `RegisteredRisingEdge`, and `RegisteredFallingEdge`.

I don't really see much of a point in adding the conditional to the call, that conditioning can be in the user's code. Adding the condition will just make it harder to port to C in the future, and there is no potential for optimization at that point. Instead of passing the conditional we could support both direct waits as well as decorator syntax:

```python

# trigger style
await RegisteredRisingEdge(dut.clk, dut.validOut)
result = dut.dataOut.value

#---------------------------

# callback decorator style
@RegisteredRisingEdge(dut.clk, dut.valid)
def read_dataOut():
    return dut.dataOut.value

#---------------------------

# ???
@Always
@RegisteredRisingEdge(dut.clk, dut.valid)
def dataOut_monitor():
    result = dut.dataOut.value
    do_thing(result)

```

eric-wieser, 2020-04-22:
The callback style can't work, as there's no `await` there.

eric-wieser, 2020-04-22:
> I don't really see much of a point in adding the conditional to the call, that conditioning can be in the user's code.

I suppose it's just a way for the user to avoid having to write a loop - they can just ask to wait until the first edge where the condition is true, and they get the loop for free.

> Adding the condition will just make it harder to port to C in the future,

Not that much harder - calling a lambda is fairly straightforward from c, and still far faster than resuming the scheduler.

ktbarrett, 2020-04-22:
@eric-wieser I don't think we can ignore the scheduler in the condition check, that's why I think it's harder. It's easy enough to say "don't do anything stupid", but the scheduler might be entered from that function and then we have another point of entry into the scheduler we have to worry about.

eric-wieser, 2020-04-22:
> but the scheduler might be entered from that function

As long as the function is not a coroutine, I think it would be safe?

chiraag, 2020-04-23:
I am sure if this is directly relevant to this discussion but I had the following code:
```py
    @cocotb.coroutine
    def read(self, num_vals):
        """ Deposit num_vals outputs respecting the rdy and en signals """
        clk_trig = RisingEdge(self.clock)
        rdy_trig = Edge(self.bus.rdy)

        vals = []
        for idx in range(num_vals):
            while True:
                trig = yield First(clk_trig, rdy_trig)
                if (
                    trig is clk_trig
                    and self.bus.rdy.value
                    and int(self.bus.en_lvl.value) <= int(self.bus.rdy_lvl.value)
                ):
                    vals[idx] = self.bus.rdy.value
                    break
                else:
                    self.bus.en <= self.bus.rdy.value
                    self.bus.en_lvl <= 4
        self.bus.en <= 0
```

This was working correctly till recently. However when running on master I see that `self.bus.en <= self.bus.rdy.value` is no longer valid. The RHS is a combinational output from the DUT and which feeds into `en` which needs to be registered on the succeeding rising edge of clock.

I do not believe I was relying on a specific ordering of the scheduler since I was explicitly checking for the right state and assigning to `en`. Is this user error or a regression in cocotb?

chiraag, 2020-04-23:
In this case the issue seems to be using the "non-blocking assign" syntax instead of `setimmediatevalue()`.  Perhaps a simple timing diagram in the documentation of when the various coroutines and triggers get scheduled would be a useful reference for developers.

themperek, 2020-04-23:
@chiraag It is interesting but can you move your findings to new issues? It does not look to me it belongs here. Also would be good to say what simulator and version you are using.

ktbarrett, 2020-04-23:
@eric-wieser The current implementation of `fork` causes an immediate `schedule`.

We could write it like what is in the OP: a coroutine that takes a conditional function; that part can stay in Python. We could encapsulate the registered edge trigger so we can move to a C impl of that trigger later and not have to worry about user code in the impl. That seems like a proper separation that still allows us to write what was originally suggested.

```python
async def wait(clk, signals, condition)
    while True:
        await First(*(RegisteredEdge(clk, signal) for signal in signals))
        if condition():
            break

async def RegisteredEdge(clk, signal):
    # move this to C later
    prev_value = signal.value
    while True:
        await RisingEdge(clk)
        value = signal.value
        if value != prev_value:
            break
        await Edge(signal) 
```

alexforencich, 2020-11-22:
You know, it would be very useful to do something along these lines to support efficient streaming interfaces - AXI, AXI lite, AXI stream, and Avalon are all based on streaming interfaces with very simple (and very similar) ready/valid handshaking.  If the handling of the handshaking can be kicked over to the C side in a nice way, then this could make interfacing with these sorts of interfaces much more efficient.  Instead of conditioned events, the idea would be to have conditioned driving and sampling, and you would just send the raw data to/from the python side.

cmarqu, 2021-01-12:
Related: Cadence's Specman/e language and tool have something they call an `event_port`: https://community.cadence.com/cadence_blogs_8/b/fv/posts/usage-of-event-ports-with-edge-attribute-to-define-simulator-sensitive-events-rather-than-simple-ports