cocotb/cocotb #1699: Problem with running endian_swapper VHDL example.

m-kru, 2020-04-23:
I have installed cocotb from pip. Followed this [Quickstart Guide](https://docs.cocotb.org/en/latest/quickstart.html) for VHDL example.
When I execute: `make SIM=ghdl TOPLEVEL_LANG=vhdl` in `cocotb/examples/endian_swapper/tests` directory the output is:
```
/home/mkru/.local/lib/python3.8/site-packages/cocotb/share/makefiles/Makefile.sim:52: *** "Couldn't find makefile for simulator: "ghdl"! Available simulators: ".  Stop.
```
Any idea what might be wrong?

themperek, 2020-04-23:
Very strange. Looks like something very wrong with the installation?

Maybe uninstall (`pip uninstall`) and install one more time?

m-kru, 2020-04-23:
@themperek Still the same. `Makefile.ghdl` is located in 
```
/home/mkru/.local/lib/python3.8/site-packages/cocotb/share/makefiles/simulators
```

themperek, 2020-04-23:
What version are you using?

eric-wieser, 2020-04-23:
Looks like something is wrong with these lines

https://github.com/cocotb/cocotb/blob/v1.3.1/cocotb/share/makefiles/Makefile.sim#L45-L53

themperek, 2020-04-23:
This can happen if you use example from master but you installed previous  version.

m-kru, 2020-04-23:
No, still the same result.

themperek, 2020-04-23:
Try another example? But this should work. I do not see what can go wrong.
Maybe ghdl version?
Do you have `work-obj93.cf` in `sim_build` directory?

m-kru, 2020-04-23:
It is the same with dff example. Yes `work-obj93.cf` is in the `sim_build`. Does cocotb require ghdl to be build with specific back end?

m-kru, 2020-04-23:
If I run
```
/usr/local/bin/ghdl -r --work=work endian_swapper_vhdl --vpi=/home/mkru/.local/lib/python3.8/site-packages/cocotb-1.4.0.dev0-py3.8-linux-x86_64.egg/cocotb/libs/libcocotbvpi_ghdl.so
```
inside `sim_build` directory the simulation starts and I get another error:
```
loading VPI module '/home/mkru/.local/lib/python3.8/site-packages/cocotb-1.4.0.dev0-py3.8-linux-x86_64.egg/cocotb/libs/libcocotbvpi_ghdl.so'
     -.--ns INFO     cocotb.gpi                         ..mbed/gpi_embed.cpp:71   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     cocotb.gpi                         ../gpi/GpiCommon.cpp:106  in gpi_print_registered_impl       VPI registered
VPI module loaded!
Traceback (most recent call last):
  File "/home/mkru/.local/lib/python3.8/site-packages/cocotb-1.4.0.dev0-py3.8-linux-x86_64.egg/cocotb/__init__.py", line 144, in _initialise_testbench
    log.info("Running tests with cocotb v%s from %s" %
AttributeError: module 'cocotb.log' has no attribute 'info'
cocotb initialization failed - exiting
Failing test at simulator request before test run completion: Simulator shutdown prematurely
```

themperek, 2020-04-23:
What GHDL version are you using?

eric-wieser, 2020-04-23:
> AttributeError: module 'cocotb.log' has no attribute 'info'

That error means that somehow `COCOTB_SIM` is not set in the environment

m-kru, 2020-04-23:
@themperek I have compiled GHDL 2-3 days ago from master branch:
```
GHDL 1.0-dev (v0.37.0-382-gde0014dd) [Dunoon edition]
 Compiled with GNAT Version: 9.3.0
 llvm code generator
Written by Tristan Gingold.
```

themperek, 2020-04-23:
So we can try to debug maybe something changed we test v0.37 release.

Can you edit this file: `/home/mkru/.local/lib/python3.8/site-packages/cocotb-1.4.0.dev0-py3.8-linux-x86_64.egg/cocotb/share/makefiles/simulators/Makefile.ghdl`

You can check if this is right with: `cocotb-config --prefix`

Remove line 68 `cd $(SIM_BUILD) &&`: https://github.com/cocotb/cocotb/blob/master/cocotb/share/makefiles/simulators/Makefile.ghdl#L68
and in line 79 remove  `--workdir=$(SIM_BUILD)` :https://github.com/cocotb/cocotb/blob/master/cocotb/share/makefiles/simulators/Makefile.ghdl#L79

m-kru, 2020-04-23:
It works with changes you have proposed.

themperek, 2020-04-23:
So it looks something has changed in laters GHDL there are potentially other solutions to this.

m-kru, 2020-04-23:
Maybe @tgingold could help resolve what is going on?

tgingold, 2020-04-23:
From memory, I don't see which ghdl commit could have broken cocotb.  Do you volunteer for a bisect ?

themperek, 2020-04-23:
I have just checked the latest ghdl master and it works. What os?

m-kru, 2020-04-23:
Linux arch 5.6.4-arch1-1 #1 SMP PREEMPT Mon, 13 Apr 2020 12:21:19 +0000 x86_64 GNU/Linux

m-kru, 2020-04-23:
@themperek what back end do you use for GHDL?

themperek, 2020-04-23:
>what back end do you use for GHDL?

No idea. Default?

m-kru, 2020-04-23:
cocotb works with GHDL compiled with the default mcode back end. However it does not work with GHDL compiled with LLVM backend. On GHDL website I have found this:
[image]
It states that the call with -r argument can be replaced with direct execution of the binary. It does not state that it MUST be replaced with direct execution. @tgingold can you comment?

themperek, 2020-04-23:
It looks like something with makefile (`cd SIM_BUILD`) or how  `--workdir=` in ghdl works. Unfortunately, I am not an expert on any.

themperek, 2020-04-23:
@m-kru 

Can you try this ? 
```
# Compilation phase
analyse: $(VHDL_SOURCES) $(SIM_BUILD) $(CUSTOM_COMPILE_DEPS)
	$(foreach SOURCES_VAR, $(filter VHDL_SOURCES_%, $(.VARIABLES)), \
		$(CMD) -i $(GHDL_ARGS) $(COMPILE_ARGS) --workdir=$(SIM_BUILD)--work=$(SOURCES_VAR:VHDL_SOURCES_%=%) $($(SOURCES_VAR)) && ) \
	$(CMD) -i $(GHDL_ARGS) $(COMPILE_ARGS) --workdir=$(SIM_BUILD) --work=$(RTL_LIBRARY) $(VHDL_SOURCES) && \
	$(CMD) -m $(GHDL_ARGS) $(COMPILE_ARGS) --workdir=$(SIM_BUILD) --work=$(RTL_LIBRARY) $(TOPLEVEL)

$(COCOTB_RESULTS_FILE): analyse $(CUSTOM_SIM_DEPS)
	-@rm -f $(COCOTB_RESULTS_FILE)

	MODULE=$(MODULE) \
        TESTCASE=$(TESTCASE) TOPLEVEL=$(TOPLEVEL) TOPLEVEL_LANG=$(TOPLEVEL_LANG) COCOTB_SIM=1 \
	$(CMD) -r $(GHDL_ARGS) --workdir=$(SIM_BUILD) --work=$(RTL_LIBRARY) $(TOPLEVEL) --vpi=$(LIB_DIR)/libcocotbvpi_ghdl.$(LIB_EXT) $(SIM_ARGS) $(PLUSARGS) && pwd

	$(call check_for_results_file)

clean::
	-@rm -rf $(SIM_BUILD)
endif
```

m-kru, 2020-04-23:
@themperek This solution works with both back ends, mcode and LLVM.

themperek, 2020-04-23:
@m-kru Can you make a PR with this? If not I will do at some point.

m-kru, 2020-04-23:
I will, but lets wait. I am curious what was the problem from GHDL point of view. I would like to see what @tgingold writes first.

tgingold, 2020-04-23:
For `ghdl -r`: it simply executes the binary so the doc is correct: `ghdl -r` can be replaced with the direct execution.

In general, the options of ghdl must be the same for -i/-m/-r, in particular `--std`, `--work` and `--workdir`.

closed 2020-04-24 by themperek

pbijoy, 2020-07-16:
/usr/local/lib/python3.6/dist-packages/cocotb/share/makefiles/Makefile.sim:52: *** "Couldn't find makefile for simulator: "icarus"! Available simulators: ".  Stop.

I have installed iverilog. but i get the above error - any help ?

themperek, 2020-07-16:
@pbijoy Normally means your examples are from different version than cocotb installation.

PR #1702 (2020-04-24): Improve Makefile.ghdl
Make Makefile for GHDL work with any GHDL back end.
Previously only mcode back end worked.

Closes #1699