cocotb/cocotb #1701: Procedure access

MansurHu, 2020-04-23:
I would like to know if it is possible to de VHDL procedure access from Cocotb.
I couldn't find this information in the Cocotb docs.

For example i would like to call the following VHDL procedure from a Cocotb test:
write_field(word, 0, 2, 10,  "000000011");

Is this possible? then I would also like to change the parameters from the write_field() within the Cocotb test

cmarqu, 2020-04-23:
This is not directly possible unfortunately.
An untested workaround would be to call the procedure from a process that is sensitive to a signal that you toggle from cocotb. You can also provide signals for the inputs/outputs of the procedure and write/read them from cocotb.

If you try this, it would be very nice if you could report back your experiences. Good luck!

cmarqu, 2020-05-08:
In case you want to report back, contact us in our gitter channel at https://gitter.im/cocotb/Lobby or open a new ticket for documentation improvement (if the suggestion worked).

closed 2020-05-08 by cmarqu