cocotb/cocotb #1703: Are dff, mean, endian_swapper examples somehow outdated?

m-kru, 2020-04-24:
All three examples use `@cocotb.coroutine` decorators. If I understand [Coroutines](https://docs.cocotb.org/en/latest/coroutines.html#coroutines) correctly, `async/await` should be used right now. Am I right?

eric-wieser, 2020-04-24:
Both will still be supported in the upcoming release.

One of the last items left in #1289 is to update these examples to use the new format. If you're looking to contribute, a patch to update even one example would be very welcome!

m-kru, 2020-04-24:
I am trying to update `dff` example, but I'm stuck trying to figure out how replacing `@cocotb.coroutine` with `async` affect on `TestFactory` usage.

eric-wieser, 2020-04-24:
You should be able to replace
```
@cocotb.coroutine
def ...
```
with
```
async def ...
```

m-kru, 2020-04-24:
That's what I did. But it results with errors:
```
make results.xml
make[1]: Entering directory '/home/mkru/workspace/cocotb/examples/dff/tests'
mkdir -p sim_build
\
/usr/local/bin/ghdl -i   --workdir=sim_build --work=work /home/mkru/workspace/cocotb/examples/dff/tests/../hdl/dff.vhdl && \
/usr/local/bin/ghdl -m   --workdir=sim_build --work=work dff
analyze /home/mkru/workspace/cocotb/examples/dff/tests/../hdl/dff.vhdl
elaborate dff
MODULE=dff_cocotb TESTCASE= TOPLEVEL=dff TOPLEVEL_LANG=vhdl COCOTB_SIM=1 \
/usr/local/bin/ghdl -r  --workdir=sim_build --work=work dff --vpi=/home/mkru/.local/lib/python3.8/site-packages/cocotb-1.4.0.dev0-py3.8-linux-x86_64.egg/cocotb/libs/libcocotbvpi_ghdl.so
loading VPI module '/home/mkru/.local/lib/python3.8/site-packages/cocotb-1.4.0.dev0-py3.8-linux-x86_64.egg/cocotb/libs/libcocotbvpi_ghdl.so'
     -.--ns INFO     cocotb.gpi                         ..mbed/gpi_embed.cpp:71   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     cocotb.gpi                         ../gpi/GpiCommon.cpp:106  in gpi_print_registered_impl       VPI registered
VPI module loaded!
     0.00ns INFO     cocotb.gpi                                gpi_embed.cpp:332  in embed_sim_init                  Running on GHDL version 0.1
     0.00ns INFO     cocotb.gpi                                gpi_embed.cpp:333  in embed_sim_init                  Python interpreter initialized and cocotb loaded!
     0.00ns INFO     cocotb                                      __init__.py:144  in _initialise_testbench           Running tests with cocotb v1.4.0.dev0 from /home/mkru/.local/lib/python3.8/site-packages/cocotb-1.4.0.dev0-py3.8-linux-x86_64.egg/cocotb
     0.00ns INFO     cocotb                                      __init__.py:162  in _initialise_testbench           Seeding Python random module with 1587721759
     0.00ns CRITICAL cocotb.regression                         regression.py:154  in initialise                      Failed to import module dff_cocotb: TestFactory requires a cocotb coroutine
     0.00ns INFO     cocotb.regression                         regression.py:155  in initialise                      MODULE variable was "dff_cocotb"
     0.00ns INFO     cocotb.regression                         regression.py:156  in initialise                      Traceback:
     0.00ns INFO     cocotb.regression                         regression.py:157  in initialise                      Traceback (most recent call last):
                                                                                                                       File "/home/mkru/.local/lib/python3.8/site-packages/cocotb-1.4.0.dev0-py3.8-linux-x86_64.egg/cocotb/regression.py", line 152, in initialise
                                                                                                                         module = _my_import(module_name)
                                                                                                                       File "/home/mkru/.local/lib/python3.8/site-packages/cocotb-1.4.0.dev0-py3.8-linux-x86_64.egg/cocotb/regression.py", line 68, in _my_import
                                                                                                                         mod = __import__(name)
                                                                                                                       File "/home/mkru/workspace/cocotb/examples/dff/tests/dff_cocotb.py", line 157, in <module>
                                                                                                                         factory = TestFactory(run_test)
                                                                                                                       File "/home/mkru/.local/lib/python3.8/site-packages/cocotb-1.4.0.dev0-py3.8-linux-x86_64.egg/cocotb/regression.py", line 593, in __init__
                                                                                                                         raise TypeError("TestFactory requires a cocotb coroutine")
                                                                                                                     TypeError: TestFactory requires a cocotb coroutine

Traceback (most recent call last):
  File "/home/mkru/.local/lib/python3.8/site-packages/cocotb-1.4.0.dev0-py3.8-linux-x86_64.egg/cocotb/__init__.py", line 181, in _initialise_testbench
    regression_manager.initialise()
  File "/home/mkru/.local/lib/python3.8/site-packages/cocotb-1.4.0.dev0-py3.8-linux-x86_64.egg/cocotb/regression.py", line 152, in initialise
    module = _my_import(module_name)
  File "/home/mkru/.local/lib/python3.8/site-packages/cocotb-1.4.0.dev0-py3.8-linux-x86_64.egg/cocotb/regression.py", line 68, in _my_import
    mod = __import__(name)
  File "/home/mkru/workspace/cocotb/examples/dff/tests/dff_cocotb.py", line 157, in <module>
    factory = TestFactory(run_test)
  File "/home/mkru/.local/lib/python3.8/site-packages/cocotb-1.4.0.dev0-py3.8-linux-x86_64.egg/cocotb/regression.py", line 593, in __init__
    raise TypeError("TestFactory requires a cocotb coroutine")
TypeError: TestFactory requires a cocotb coroutine
     0.00ns ERROR    cocotb.gpi                                gpi_embed.cpp:419  in embed_sim_init                  cocotb initialization failed - exiting
[... cut ...] ompletion: Simulator shutdown prematurely
ERROR: results.xml was not written by the simulation!
make[1]: *** [/home/mkru/.local/lib/python3.8/site-packages/cocotb-1.4.0.dev0-py3.8-linux-x86_64.egg/cocotb/share/makefiles/simulators/Makefile.ghdl:76: results.xml] Error 1
make[1]: Leaving directory '/home/mkru/workspace/cocotb/examples/dff/tests'
make: *** [/home/mkru/.local/lib/python3.8/site-packages/cocotb-1.4.0.dev0-py3.8-linux-x86_64.egg/cocotb/share/makefiles/Makefile.inc:40: sim] Error 2
```

eric-wieser, 2020-04-24:
Good catch, that's a bug. The check should be relaxed to allow async functions too.
cc @ktbarrett, who might have already filed something about this.

In the meantime, you can use
```python
@cocotb.coroutine
async def a_test_for_using_with_TestFactory
```

ktbarrett, 2020-04-24:
@m-kru Thanks for taking time to update some of the examples!

The issue with `TestFactory` is here
https://github.com/cocotb/cocotb/blob/737edca64625803b19e24dd2c9ecddca99c90d28/cocotb/regression.py#L592-L593

An easy solution is to change this to
```python
        if not isinstance(test_function, cocotb.coroutine) or inspect.iscoroutinefunction(test_function):
            raise TypeError("TestFactory requires a cocotb coroutine")
```

m-kru, 2020-04-24:
@ktbarrett do you want me to update this or you will do it yourself?

ktbarrett, 2020-04-24:
You can put that change in as a separate commit in your example update PR, it is not a pressing issue for anyone else right now.

m-kru, 2020-04-24:
Something weird happens. If you run below python script:
```python
import inspect

async def a():
    pass

if inspect.iscoroutinefunction(a):
    print("True")
else:
    print("False")
```
The result is True. However if you apply this patch:
```diff
diff --git a/examples/dff/tests/dff_cocotb.py b/examples/dff/tests/dff_cocotb.py
index 328e15a3..5a9f9fdb 100644
--- a/examples/dff/tests/dff_cocotb.py
+++ b/examples/dff/tests/dff_cocotb.py
@@ -130,8 +130,7 @@ class DFF_TB(object):
         self.stopped = True

-@cocotb.coroutine
-def run_test(dut):
+async def run_test(dut):
     """Setup testbench and run a test."""

     cocotb.fork(Clock(dut.c, 10, 'us').start(start_high=False))
@@ -153,6 +152,12 @@ def run_test(dut):
     # Print result of scoreboard.
     raise tb.scoreboard.result

+import inspect
+from time import sleep
+if inspect.iscoroutinefunction(run_test):
+    sleep(1)
+else:
+    sleep(5)

 # Register the test.
 factory = TestFactory(run_test)
```
you will notice 5 seconds delay. As if the `run_test` function was not recognized as coroutine.

m-kru, 2020-04-24:
Something weird happens. If you run following script:
```python
import inspect

async def a():
    pass

if inspect.iscoroutinefunction(a):
    print("True")
else:
    print("False")
```
The result will be True. However when you apply following parch:
```diff
diff --git a/examples/dff/tests/dff_cocotb.py b/examples/dff/tests/dff_cocotb.py
index 328e15a3..5a9f9fdb 100644
--- a/examples/dff/tests/dff_cocotb.py
+++ b/examples/dff/tests/dff_cocotb.py
@@ -130,8 +130,7 @@ class DFF_TB(object):
         self.stopped = True

-@cocotb.coroutine
-def run_test(dut):
+async def run_test(dut):
     """Setup testbench and run a test."""

     cocotb.fork(Clock(dut.c, 10, 'us').start(start_high=False))
@@ -153,6 +152,12 @@ def run_test(dut):
     # Print result of scoreboard.
     raise tb.scoreboard.result

+import inspect
+from time import sleep
+if inspect.iscoroutinefunction(run_test):
+    sleep(1)
+else:
+    sleep(5)

 # Register the test.
 factory = TestFactory(run_test)
```
you will notice 5 seconds delay. As if `run_test` was not recognized as coroutine.

eric-wieser, 2020-04-24:
Can you do something more decisive like `print` or `assert` to test that?

m-kru, 2020-04-25:
[image]
```diff
diff --git a/examples/dff/tests/dff_cocotb.py b/examples/dff/tests/dff_cocotb.py
index 328e15a3..fa871780 100644
--- a/examples/dff/tests/dff_cocotb.py
+++ b/examples/dff/tests/dff_cocotb.py
@@ -130,8 +130,7 @@ class DFF_TB(object):
         self.stopped = True

-@cocotb.coroutine
-def run_test(dut):
+async def run_test(dut):
     """Setup testbench and run a test."""

     cocotb.fork(Clock(dut.c, 10, 'us').start(start_high=False))
@@ -153,6 +152,8 @@ def run_test(dut):
     # Print result of scoreboard.
     raise tb.scoreboard.result

+import inspect
+assert inspect.iscoroutinefunction(run_test)

 # Register the test.
 factory = TestFactory(run_test)
```
[image]

eric-wieser, 2020-04-25:
Can you use `assert inspect.iscoroutinefunction(run_test), "got {!r} (type {!r})".format(run_test, type(run_test)"`?

m-kru, 2020-04-25:
```
assert inspect.iscoroutinefunction(run_test), "got {!r} (type {!r})".format(run_test, type(run_test))
AssertionError: got <function run_test at 0x7fe6d7353430> (type <class 'function'>)
```

eric-wieser, 2020-04-25:
Can you print `sys.version_info` from within cocotb too?

m-kru, 2020-04-25:
```
sys.version_info(major=3, minor=8, micro=2, releaselevel='final', serial=0)
```

eric-wieser, 2020-05-03:
I have no idea what's going wrong with your setup. Can you make the patch anyway, and see if our CI is happy with it?

m-kru, 2020-05-03:
https://github.com/cocotb/cocotb/pull/1752

eric-wieser, 2020-06-23:
Let's close this in favor of gh-1730

closed 2020-06-23 by eric-wieser